RavenSystem/esp-homekit-devices #1447: Sonoff Mini R2 GPIO2 issues

iondarie, 2021-06-12:
Hello,

I am using a sonoff mini r2 to operate my garage door which has 2 reed sensor one for fully open one for fully closed.

My closed reed sensor is connected to GPIO2, when the garage is closed the mini refuses to boot because it is being pulled low by the reed sensor, Is there any way to set GPIO2 to be pulled high on boot?

Here is the code i am using:
```
{
  "c": {
    "l": 13,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12,
            "v": 1,
            "i": 0.5
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1,
            "i": 0.5
          }
        ]
      },
      "2": {
        "r": []
      },
      "3": {
        "a": 2
      },
      "t": 40,
      "d": 43,
      "f3": [
        {
          "g": 2,
          "t": 0
        }
      ],
      "f4": [
        {
          "g": 2,
          "t": 1
        }
      ],
      "f2": [
        {
          "g": 4,
          "t": 0
        }
      ],
      "f5": [
        {
          "g": 4,
          "t": 1
        }
      ],
      "s": 5,
      "b": []
    }
  ]
}
```

mariusvasss, 2021-06-21:
Incearca sa-l pui pe pinul de ota, gpio 16.

iondarie, 2021-06-23:
@mariusvasss multumesc dar din pacate Sonoff Mini R2 nu mai are GPIO 16...

mariusvasss, 2021-06-23:
Nici pe rx nu merge? E GPIO3 daca nu ma insel. Eu am pus senzor de temperatura pe tx-ul din sonoff touch si merge perfect.

iondarie, 2021-06-23:
@mariusvasss am pus si pe RX si pe TX dar dupa un timp se prajeste si nu mai vrea sa vada senzorul

mariusvasss, 2021-06-23:
Hmmm, atunci cu putina maiestrie si rabdare, lipeste-te de 14 pe chip, tre' sa mearga. @iondarie

mariusvasss, 2021-06-23:
Eu folosesc NC pe 4 si 2 iar 2 il pun sa indice open. @iondarie

closed 2021-07-24 by stale[bot]