RavenSystem/esp-homekit-devices #1451: D1 mini + 4 relay + remote to control garage door. How to code Json config?

MJdinkel, 2021-06-24:
Hi, 
It’s my first time to DIY a smart device. I use D1 mini + 4 relay + remote (as the image) to flash Tasmota and upgrade HAA. I have no idea about Jason config. I think this ([haa-configurator](https://glumb.github.io/haa-configurator/) ) might be helpful.  But I don’t know how use. 
[image]

MJdinkel, 2021-06-26:
Can I use this Jason code?

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"i":0.5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]},{"i":0.5,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"i":0.5,"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}]},{"i":0.5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}]}]}

closed 2021-07-13 by MJdinkel

ch0w7, 2023-11-08:
did you figure out the code? I have a d1 mini and a 4 relay and looking to get HAA configuration