RavenSystem/esp-homekit-devices #1453: MagicHome rgb stop working

mafyata, 2021-06-27:
Hi, my rgb controller stop working last night. The controller is connected to wifi but in my router settings is shown as haa-f8f4fc, not the name I entered in json. Now it is not shown in home app. I tried reset settings, reset homekit id , but the problem is same. 
[image]
[image]
[image]

mafyata, 2021-06-27:
This is my json that worked until last night. 
{"c":{"n":"Hol-Led","b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13,"fr":5,"fg":5,"fv":5},{"0":{"r":[]},"1":{"r":[],"s":[{"a":1}]},"t":1,"b":[],"s":0}]}

Hanimaniac, 2021-06-27:
I had an similar Issue, while updating to the latest Version.
I had had to change the rgb gpio lines in the jason from "r":5,"g":12,"v":13 to "g":[5,12,13]. This is a newer confic scheme from HAA. After that Change, my Controller worked again (but was shown in home all the time, maybe you have to change some other parts of your json).

mafyata, 2021-06-27:
Can someone help with new scheme.

mafyata, 2021-06-27:
Everything is ok now. Thanks.

closed 2021-07-27 by stale[bot]