RavenSystem/esp-homekit-devices #1456: Wemos D1 Mini not working after recent update

bencharlesrhys, 2021-06-30:
I recently updated to the latest version on my Wemos D1 Mini and the GPIO pins I set up stopped working. I reflashed the device with latest HAA and I can see the HAA-D12322 wifi network and everything but can't get onto the setup server at 192.168.4.1. Config below:

`{
  "c": {
    "l": 14,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ],
    "n": "Coffee-Machine"
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 4,
            "v": 1,
            "i": 0.5
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 4,
            "v": 1,
            "i": 0.5
          }
        ]
      },
      "t": 1,
      "b": [],
      "s": 0,
      "i": 1200
    }
  ]
}
`

xrust83, 2021-07-01:
Read the wiki!  to connect the latest versions, 192.168.4.1 is not enough, you also need to specify the port 192.168.4.1:4567 and this information is indicated in the first two paragraphs about the installation

bencharlesrhys, 2021-07-01:
Thank you muchly!

closed 2021-07-01 by bencharlesrhys