RavenSystem/esp-homekit-devices #1457: Are the new Sonoff devices compatible with HAA (not actually an issue)

tlagarto, 2021-07-01:
This is not actually an issue but I wanted to know if these devices are compatible with HAA fw:
SONOFF DUALR3
SONOFF TX Series
SONOFF MINIR2
SONOFF T4EU1C

Could some one help?
Thanks in advance

mariusvasss, 2021-07-01:
Mini r2 is, tested!

mariusvasss, 2021-07-04:
@tlagarto if you mean the touch wall switch, tested all 1,2,3 gangs.

Murdrock, 2021-07-07:
how about SONOFF DUALR3 ? Will there be any support soon?

Thanks in advance.

mariusvasss, 2021-07-07:
If it's an esp will work and I think it is.

Murdrock, 2021-07-07:
I have a couple of em coming. I'll make a test. But I see that it is ESP32. Will I be able to firmware it?

mikwayhk, 2021-07-21:
> @tlagarto if you mean the touch wall switch, tested all 1,2,3 gangs.

I'm currently using the T2UK version, one 1 gang & two 3 gangs. Could u tell me how to flash HAA into the chips? Cuz i have trouble to find which spot connecting PC....

closed 2021-08-21 by stale[bot]