RavenSystem/esp-homekit-devices #1459: Often unresponsive

775728212, 2021-07-03:
Is frequent non-response a router problem

peros550, 2021-07-03:
It could be the router. I have more than 10 devices in my network and only 1 or 2 go unresponsive very frequently.Always those two.  I believe it has to do with the device rather than the firmware.

Daveblanche, 2021-07-03:
There is a pave somewhere that talks about apples recommendation on router setup for HomeKit. I found it when I was having issues. Turned out to be meshed based systems in our (relatively) big and awkwardly shaped house. Will look it up later, if I can find it. Was very useful. It worked. 
But in the end I got all fancy and I stalked a UniFi prosumer system throughout my house. ..

Daveblanche, 2021-07-04:
Can’t find it sorry. I seem to remember  these points, but there may be more. Same or similar settings based on different router. I found useful advice on asus and unifi forums and now have zero issues. (For now) I moved from asus to unifi. 
mesh was a major issue in my case, but not everyone suffers from that and I’m sure it depends on the system. 
mdns on, might only be required if running vlans 
Fast roaming off
Igmp snooping on
Igmpv3 on
Dtim Mode 3 
Basically, anything else that adds artificial improvement of service that should be run by the client connecting to the router and not the router itself could be contributing.
So try any WiFi ai off
Any auto optimisation off
And see how that goes
But it is probably very router specific.

duchenpaul, 2021-07-05:
Often unresponsive --  This happens to me as well, I am new to HAA, I used sonoff, and if I keep toggling sonoff in a short time period, (keep tapping on/off), it becomes dead after (keep toggling) 4 seconds, can be recovery by rebooting device.
I did the same using eshome home + home assistant, with the same hardware and actions, no such issue.

RavenSystem, 2021-07-05:
> Often unresponsive -- This happens to me as well, I am new to HAA, I used sonoff, and if I keep toggling sonoff in a short time period, (keep tapping on/off), it becomes dead after (keep toggling) 4 [...]

You are entering setup mode: https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#entering-setup-mode

xrust83, 2021-07-05:
you are lazy and do not want to read the wiki.  Quickly switching the haa device with a physical button, or from the home application, causes the haa device to go into configuration mode.  and this is normal, since in ordinary life, a person does not need to turn on and off the light 4 times in 5 seconds

closed 2021-08-04 by stale[bot]