RavenSystem/esp-homekit-devices #1466: Grarage Door without any Sensors

Unwies, 2021-07-06:
Hey, I want to use a Sonoff Basic R2 as a trigger for my garage door. The relays will only close for some seconds and this is connected parallel to a open/close button.

So far this works fine as a switch:
{"c":{"o":1,"l":13,"i":0},"a":[{"t":1,"i":2,"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]}}]}

but, I would like to make this a Type 40 (garage door) to be presented in Apple Car Play as a Garage Door.

I tried to create a configuration without any sensor inputs, but I failed...

Is it possible to have such a config without any sensor input?

Failed:
{
  "c":{"l":13,"b":[{"g":0,"t":5}]},
  "a":[{
    "t":40,
    "d":2,
    "0":{"r":[{"g":12,"v":1,"i":2}]},
    "1":{"a":0}
  }]
}

mariusvasss, 2021-07-06:
@Unwies 

{
  "c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},
  "a":[{
    "t":40,
    "d":18,
    "0":{"r":[{"g":12,"v":1,"i":0.5}]},
    "1":{"a":0},
    "2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},
    "3":{"a":2}
  }]
}

mariusvasss, 2021-07-06:
@Unwies Try this. 
Not recommend without sensors but if you need a configuration like this, you have it. It will not report the status of your door when you push the button in parallel.

Unwies, 2021-07-06:
@mariusvasss  many thanks - works perfectly

mariusvasss, 2021-07-06:
You welcome @Unwies

closed 2021-08-07 by stale[bot]