RavenSystem/esp-homekit-devices #1467: Light Sensor (BH1750)

andrewbpowers, 2021-07-09:
Hello!

The chapter about light sensors is quite little. Is it possible to use a BH1750 sensor?
Because for this you need two GPIOs.

Thanks for any tips!

Take care...

lwqcz, 2021-08-05:
Hi,
sensor BH1750 is I2C bus digital one. Currect support in HAA code is ready for LDR based analog ones only, typically located inside PIR motion sensors to prevent PIR, used typically with light bulb, turn output on during day.

Anyway - `LIGHT_SENSOR_TYPE` could be modified to accept another sensor type and deeper in code, instead on ADC RAW read, there would have be a I2C communication with sensor (it is far more complex, then LDR).

Support for digital light sensors would be useful IMHO.

andrewbpowers, 2021-08-05:
Thank you very much for your detailed answer!

I am now running the sensor and a DHT22 together with the Tasmota firmware.
Including the integration with Home Assistant it was a matter of only a few minutes.
Everything runs without problems.

closed 2021-08-22 by RavenSystem