RavenSystem/esp-homekit-devices #1469: Add support for Shelly 1/1PM sensor-addon

tunip, 2021-07-10:
Would be great if support for the Shelly 1/1PM sensor-addon could be added into HAA.

Why integrate this? For safety reasons, because the sensor DS18x20 is galvanically isolated from the Shelly and 230V itself. I would like to build thermostats based on Shelly 1 with this sensor-addon. But right now, it is not supported in HAA. And directly connecting the DS18x20 sensor (mentioned in the "Tested JSON Collection") is danger and not an option!

If you need a Shelly 1 with this sensor-addon I will donate one to you for help and support your great work.

Sensor-Addon:
https://shelly.digital/shop/accessories/temperature-sensor-addon-for-shelly-1-1pm/

Tasmota integration:
https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_05_ds18x20.ino

Tasmota discussion:
https://github.com/arendst/Tasmota/issues/7469

tunip, 2021-07-12:
And the ESPHome integration for the Shelly sensor-addon:
https://github.com/esphome/esphome/pull/1346

kurtwarwick-new, 2021-07-27:
This would be an amazing addition to HAA! I have only 1 Shelly left being used through HomeBridge, which is the one with the sensor.

@tunip, why, might I ask, is it dangerous to use the JSON posted in the collection? I have tried it, and it just didn’t work, but I am interested to know why it’s dangerous.

Thanks

tunip, 2021-07-27:
@kurtwarwick-new
Sorry, I was not precise enough in my first post.

You can run a sensor on the Shelly 1 directly on the GPIOs like:
https://github.com/RavenSystem/esp-homekit-devices/issues/689#issuecomment-681016125

I have tested this (only with 12 volt) and it works so far.

At 230 volts, I would not run a sensor directly to the Shellys GPIOs, because GND is on phase (230 volts!).

And for my application (heating thermostat) only 230 volts are available for power supply. Thats why I need the Shelly addon to use a temperature sensor.

kurtwarwick-new, 2021-07-27:
Thanks for that explanation @tunip .

I am using the official Shelly Add On, so surely that should be constructed and connected in a way that is safe on 230v, given that the default for a Shelly 1 is 230v. It would be be a poor move if that were not the fact.

I have been running the add on with 240v with the official firmware with great results, but switching to HAA OTA just shows 0 degrees. Not sure why that would be…

Anyways, I would still like to see this move forward.

Thanks

RavenSystem, 2021-08-22:
Receiving UART info is not supported.
https://github.com/RavenSystem/esp-homekit-devices/issues/915

closed 2021-08-22 by RavenSystem

pilot1981, 2023-09-30:
It is not supported in last version?