RavenSystem/esp-homekit-devices #1471: HAA Manager

cirrusflyer, 2021-07-12:
Should I be able to update the firmware on my devices using the HAA Manager?  When try, nothing happens.  Is there another way to update to the latest version?  Currently at 5.0.18 firmware.

Daveblanche, 2021-07-13:
There is the standard push of the bottom on the unit for 8seconds or turn the unit on and off 8 times. You can also try powering it on then off (form the power source ) within (?) 5seconds to make it go into setup mode. Then use <ip.address>:4567 to look for the unit once it is in setup mode. Or you can use „radar“ in the manager App to highlight any units that are in set up mode. 
that lot should keep you busy.

cirrusflyer, 2021-07-13:
Thanks.  I can get them into setup, but I don't recall a way to update them to the latest version once there.

Daveblanche, 2021-07-13:
tick the bock next to "update all" and then click save. As long as you are viewing it through your wife network then it will update. If you are viewing it through its own Hotspot then you will have to input the correct wifi credentials first.

xrust83, 2021-07-13:
> Thanks.  I can get them into setup, but I don't recall a way to update them to the latest version once there.

[image]

cirrusflyer, 2021-07-13:
Ah yes, I recall now the last time I tried that and it made no difference.  I'll try again now.

closed 2021-08-13 by stale[bot]

krzysztofKubiakVizlib, 2022-02-06:
Did you solve your issue @cirrusflyer ?

cirrusflyer, 2022-02-07:
Yes, thanks.  The firmware updated recently with  the last two versions.  9.x and 10.x.