RavenSystem/esp-homekit-devices #1473: relay4 trigger once when device power on

justlight220, 2021-07-16:
Hi,
after flash fullhaaboot.bin firmware, relay4 not tigger when device power on
but, after setting up wifi , relay4 tigger once when device power on.
is any solution can fix it?

Device: Sonoff 4ch pro r3
Json Config
```json
{
    "c": {
        "l": 13,
        "b": [
            {
                "g": 0,
                "t": 5
            }
        ]
    },
    "a": [
        {
            "i": 0.5,
            "0": {
                "r": [
                    {
                        "g": 15
                    }
                ]
            },
            "1": {
                "r": [
                    {
                        "g": 15,
                        "v": 1
                    }
                ]
            },
            "b": [
                {
                    "g": 14
                }
            ]
        },
        {
            "i": 0.5,
            "0": {
                "r": [
                    {
                        "g": 12
                    }
                ]
            },
            "1": {
                "r": [
                    {
                        "g": 12,
                        "v": 1
                    }
                ]
            },
            "b": [
                {
                    "g": 0
                }
            ]
        },
        {
            "i": 0.5,
            "0": {
                "r": [
                    {
                        "g": 5
                    }
                ]
            },
            "1": {
                "r": [
                    {
                        "g": 5,
                        "v": 1
                    }
                ]
            },
            "b": [
                {
                    "g": 9
                }
            ]
        },
        {
            "i": 0.5,
            "0": {
                "r": [
                    {
                        "g": 4
                    }
                ]
            },
            "1": {
                "r": [
                    {
                        "g": 4,
                        "v": 1
                    }
                ]
            },
            "b": [
                {
                    "g": 10
                }
            ]
        }
    ]
}
```

justlight220, 2021-07-16:
device power on video
https://user-images.githubusercontent.com/75196944/125927185-6fbb224d-2e0e-43f1-ad0f-f233da2c587c.mov

RavenSystem, 2021-07-16:
Which GPIO uses relay 4?

justlight220, 2021-07-16:
relay use GPIO 15 
button use GPIO 14 

[image]

RavenSystem, 2021-07-16:
Let me some minutes to rebuild a beta version.

Daveblanche, 2021-07-16:
Awesome!

RavenSystem, 2021-07-16:
Try beta v6.53.1:

https://github.com/RavenSystem/haabeta

justlight220, 2021-07-16:
I'm not sure update success or not
update use these step
1. change Custom server for updates: to github.com/RavenSystem/haabeta/releases/download/6.53.1
2. change port to 443
3. enable Secure connection
4. selected Update all now (HAA OTA v4.9.0)
5. click save
6. screen show ok

setup mode still show Home Accessory Architect v6.3.0
what other step need to do?
[image]

RavenSystem, 2021-07-16:
Change Custom server for updates to: github.com/RavenSystem/haabeta/releases/latest/download

And do other steps again.

justlight220, 2021-07-16:
It works. Thank for your help.

closed 2021-07-16 by justlight220

RavenSystem, 2021-07-16:
Can you try last BETA version 6.54.0, please?

justlight220, 2021-07-16:
Also work. relay4 not trigger when power on.
[image]

RavenSystem, 2021-07-16:
Many thanks. 

You can leave beta repo removing all content of "Custom server for updates".

HAA v6.4.0 has been released including changes about this matter.

justlight220, 2021-07-17:
Got it. Thank you for your help.