RavenSystem/esp-homekit-devices #1475: Example for Security System

toshibochan, 2021-07-19:
Have examples for Security System for v6.5.1?

tunip, 2021-07-19:
In HAA Manager you have under Extra Services a "Power State" button. If this Button is "pressed", then the alarm is triggered (if system is armed). Next thing we need to find out is, how we can trigger this "Power State" button with an automation.

tunip, 2021-07-19:
The example in the Wiki is not correct. To expose the trigger switch to the Home app you need the following json config:
```
{
  "a": [{
    "t": 55,
    "es":[{
      "i": 1,
      "1": { "m": [ [ 1, 4 ] ] }
    }]
  }]
}
```
(Removed the part `"h": 2,`)

Then remove the device and add it again.

toshibochan, 2021-07-19:
Can give us examples in json

tunip, 2021-07-19:
See my "easy testing in the Home app". Use my posted json above...

CryptonWind, 2021-08-01:
> Can give us examples in json

{"c":{"b":[{"g":0,"t":5}],"l":13},"a":[{"3":{"m":[[3,0]]},"4":{"m":[[3,1]]},"6":{"m":[[3,1]]},"7":{"m":[[3,1]]},"t":55,"n":[1,2,3],"s":5,"es":[{"1":{"m":[[1,4]]},"i":1},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"t":1,"b":[{"g":0}],"s":5,"i":3600},{"1":{"m":[[1,6],[1,7]]},"t":12,"s":0,"i":180,"f1":[{"g":4,"t":1,"i":1}]},{"1":{"m":[[1,6]]},"t":12,"s":0,"i":180,"f1":[{"g":5,"t":1,"i":1}]},{"0":{"r":[]},"1":{"m":[[1,6],[1,7]]},"t":5,"s":0,"f1":[{"g":14,"t":1,"i":0}],"f0":[{"g":14,"t":0}]}]}]}

closed 2021-08-13 by RavenSystem