RavenSystem/esp-homekit-devices #1477: ESP8266 boot loop -  ets Jan  8 2013,rst cause:2, boot mode:(3,6)

hennessyevan, 2021-07-20:
I'm getting a boot loop on the current version of HAA installing on an ESP8266

```bash
Update settings

Rebooting...

del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeff68, prio:14, stack:512
pm_task_hdl : 3ffef728, prio:1, stack:176
frc2_timer_ta
```

__^ It actually stops output just like that `frc2_timer_ta`__

For sanity's sake I tried the basic lightbulb example:
```json
{
  "c":{"q":500,"b":[{"g":0,"t":5}]},
  "a":[{
    "t":30,
    "g":[15,13,12],
    "b":[{"g":0}]
  }]
}
```

icepie, 2021-08-01:
I get the same error

```log

* HAAMAIN v6.8.4 installed

Restarting...
rm match
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeff68, prio:14, stack:512
pm_task_hdl : 3ffef728, prio:1, stack:176
frc2_timer_ta␀␀␀␀␀␀␀
```

hennessyevan, 2021-08-10:
So weird.. have you used this library before @icepie?

This is the first time that I am coming across this. I'm not sure what it is. I tried a very simple configuration as well as at least 3 different chips all resulting in this same error...

Any ideas here @RavenSystem?

RavenSystem, 2021-08-10:
First, enable logs to know what is happening.

closed 2021-08-22 by RavenSystem