RavenSystem/esp-homekit-devices #1478: Can’t get hotspot captive screen

toddc01b, 2021-07-22:
I've loaded a few devices over the last year with no issues but the latest one which is a duel relay module containing an ESP-12F doesn't display a captive wifi portal in order to enter my home wifi details and other setting. 
 I used nodeMCU flasher to load 6.2 firmware. After rebooting I can see the HAAxxxxxxx wifi. I select it and the computer connects but I don't get the pop up window. I've tried multiple times and different computers but still no success. 
Any help would be appreciated. 
Thanks

tonysprenk, 2021-07-22:
Connect to the hotspot and in your browser go to http://192.168.4.1:4567/

toddc01b, 2021-07-22:
Thanks  so much. That worked

closed 2021-07-22 by RavenSystem