RavenSystem/esp-homekit-devices #1482: MagicHome LED Controller

lfiocco, 2021-07-23:
Tried flashing this to MagicHome controller. RGB, unable to load the json file. falls back to install config webpage.

Able to join network, can't load json

{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"g":[5,12,13],"b":[{"g":0}],"s":0,"p":1024,"d":1,"e":10,"fr":1,"fg":1,"fv":1}]}

was working in the past, broken since update.

toshibochan, 2021-07-23:
{"c": {"b": [{"g": 0,"t": 5}]},"a": [{"t": 30,"g": [5, 12, 13],"b": [{"g": 0}],"f2": [{"g": 4}],"f3": [{"g": 3}]}]}

I add a button on GPIO 0, GPIO 4 and GPIO 3

"b": [{"g": 0}] is for turn on and off 
"f2": [{"g": 4}] is for dimmer up 
"f3": [{"g": 3}] is for dimmer down

toshibochan, 2021-07-28:
{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"g":[5,12,13],"b":[{"g":0}]}]}

toshibochan, 2021-07-28:
Device database 
https://github.com/RavenSystem/esp-homekit-devices/wiki/Devices-Database

toshibochan, 2021-07-28:
What this part do? "s":0,"p":1024,"d":1,"e":10,"fr":1,"fg":1,"fv":1}]}

lfiocco, 2021-07-29:
Thank you. I tried your JSON seems ok now. I took the config from the HAA configurator found on the web. Yours seems to have done the job.

Thank you for your help.

toshibochan, 2021-07-29:
If is working close this issue

closed 2021-07-29 by lfiocco