RavenSystem/esp-homekit-devices #1488: Upgrade RavenCore 1.2 to HAA

rickdhoy, 2021-07-26:
Hi All,

I have been reading and reading everything I can find and feel really stupid.

I am trying to update my Ravencore devices to HAA but for the life of me I can not find the set up menu showing the ability to do the update in the EVE app. I have even downloaded an older version of the EVE App but for the life of me I can not find how to enter set up !!

I have been trying to follow these instructions:

https://github.com/RavenSystem/esp-homekit-devices/issues/686

To do the update.

For example I have not been able to find the window in the image below.

[image]

I have paid and downloaded the HAA Manager from the developer but that won't also let me update the devices.

You can see a few opt the devices below.

[image]

I am using the latest IOS and have tried both the EVE Legacy App and the latest version from the App store.

I am sure thuis is simple and I will kick myself but really need some help.

tonysprenk, 2021-07-26:
If I’m not mistaken OTA update of older Ravenscore devices is only possible when they were originally flashed with LCM OTA. It looks like your device wasn’t. The only way to update it would be to flash it by wire.

xrust83, 2021-07-26:
You are right, most likely you will not be able to just update and there are at least 3 reasons for this

1. Upgrading from ravencore to haa worked for a short period of time about 1.5 months.  a special repository was opened for this.

2. haa manager application works with version 5.0.18 if I'm not mistaken that it doesn't suit you either.

3.you could flash your devices to a version without OTA or (the last swan) that do not support updating through the air.

the best advice is to remove your devices and flash them again via the cable to the latest firmware and in the future to update the devices to the latest versions.

i3laze, 2021-07-26:
Sorry, but the instruction you’ve used ends with he words “GitHub OTA issues”.

Out of nowhere 1.5 months after HAA initial release GitHub proxy-servers hosted at Amazon introduced case-sensitive HTML Headers (specifically “Location”), which was totally incompatible with existing legacy RavenCore and HAA code.

You have to re-flash by wire.

rickdhoy, 2021-07-27:
Hi All, Thanks for the quick response. This is going to era  really painful process as I have so many of these devices :-) Ohh well.

Thanks

i3laze, 2021-07-27:
Not to mention you’ll have to learn HAA JSON syntax using Wiki.
I strongly suggest you begin with web-configurator:
https://glumb.github.io/haa-configurator/

closed 2021-08-22 by RavenSystem