RavenSystem/esp-homekit-devices #1489: Couldn't connect to accessory error on iOS 15

clemencov, 2021-07-28:
After updating on iOS 15 (Public Beta 3) I'm not able to add accessories anymore (tried on 4 different Sonoffs Mini R2), but on my wife's iPhone with iOS 14 still works without an issue.

Maybe this is not related to firmware, but reporting this, because for a couple of days wasn't able to find the root cause.

technoo10201, 2021-07-28:
Hello @clemencov a friend of mine has the same bug with his iPhone XR (iOS 15 - Beta 4) since this afternoon.

After the update of the iPhone firmware he is not able to add a new device on Homekit, but it is still possible with his wife's iPhone (iPhone SE 2020 - iOS 14.7.1).

After adding the new device from the iPhone iOS 14.7.1, he was able to see the new device on iOS 15 - Beta 4.

Hope it will be fixed soon !
Regards

ooomakon, 2021-07-29:
I have this happening with IOS 15.2

RavenSystem, 2021-08-13:
HAA doesn't support iOS beta versions. It will be fixed if issue keeps when final version is released.

closed 2021-08-13 by RavenSystem