RavenSystem/esp-homekit-devices #1492: Auto-Closing Car Gate/Garage Door

raddi, 2021-08-03:
Hey!
I want to control an automatic car gate / garage door, which opens on button press (relay on Shelly 1 switching on for 1sek) and automatically closes again after 2 minutes without any other button press.

So i want the relay to switch on for 1 sec for opening and then in the Home App the open state should change to the closed state after 2min without any other action. Is that possible?

I experimented with the following json string:

{
"c": {{"l":13,"b":[{"g":0,"t":5}]},
"a": [{
"1": {"r": [{"g": 12,"v": 1,"i": 1}]},
"0":{"r":[{"g":12,"v":0}]},
"t": 40,
"s": 0,
"d": 18
}]
}

but i dont know where to set the state to closed again after 2 mins.

so basically what i want: 
click the Garage icon in Homekit, relay switches on for 1sek, the Garage icon shows "opened", wait 120 sek, the Garage icon shows "closed", so when i click on it again, it will open the garage again (switch relay on for 1sek)

I dont want a reed-switch to monitor the open/close state because the gate will always close again automatically after 120 sek (when there is no obstruction)

raddi, 2021-08-03:
workaround 1:

make an automation in the Home app for the garage door when it opens and convert it to shortcut where you can set a wait time for closing the garage door again

workaround 2:

{"a":[{"d":120,"0":{"m":[{"g":4,"v":1}]}},{"1":{"r":[{"g":4,"v":1,"i":1}]},"0":{"r":[{"g":4,"v":0}]},"t":40,"d":5}]}

this json creates the garage door and a switch which turns off after 120 sec (d:120), then you can create 2 automations: turn switch on when garage door opens and close garage door when switch turns off. With that the switch should close the garage door after the 120 sec timer

Is there a possibility to do that without any of the Home App automation natively in HAA?

xrust83, 2021-08-03:
I asked myself and answered myself :-).  Think three times over how to do this kind of automation.

 1. Do you have children?  a child can run out or stand in the aisle of the gate.  
 2. The car can stall in the driveway, but your automation doesn't care.  
 3. Foreign objects can remain in the driveway, 2 or 3 cars can jam into the crank and a traffic jam can form. 
 
 Automation must be done wisely and protective conditions must be thought out.  at least a sensor that sees if someone or something is in the driveway

raddi, 2021-08-03:
The car gate has its own controller and sure does stop when a car or any other "thing" stands inside the door. I just want to automate the opening because the RF remote which comes with the gate doesn't have the best reception :) 
Basically i want to automate the button press on the remote control. And because the gate closes on its own, i need the homekit garage button to turn off after some time too :)

macjl, 2021-08-03:
You can use an accessory notifications to turn on a second switch (hidden from HomeKit) when garage door opens, then a delay in that switch to turn it off after 2 minutes, and then another  accessory notification to close the garage door.

i3laze, 2021-08-19:
True, internal notifications rock.
My gate works even better:
It closes either after 5 min when inching switch goes off, or 15 sec after photoelectronic motion sensor triggered.

closed 2021-08-22 by RavenSystem