RavenSystem/esp-homekit-devices #1494: Water meter support

lwqcz, 2021-08-05:
Hi,

am I correct, that HomeKit does not have support for sensors like water meter (or energy meter) is?

The input is asimple pulse on GPIO PIN, made by reed contact (in case of my type of water meter) or made by optocoupler like 4N35 driven by LDR on his input, switched by LED on the utility power meter.

I plan to use for my house [this sensor type supported by esphome](https://esphome.io/components/sensor/pulse_counter.html), plus integrate data to my Hass.io, but sometimes I would love to have a simpler Homekit solution.

Any thoughts? Thx.

tonysprenk, 2021-08-05:
Water meter is only supported as part of the humidifier accessory in HomeKit (for alerts when the container is full or empty). Energy meter is supported by HomeKit but not by the Home app. You can use an app like Eve to view energy consumption.

RavenSystem, 2021-08-06:
https://github.com/RavenSystem/esp-homekit-devices/wiki#issues

closed 2021-08-06 by RavenSystem