RavenSystem/esp-homekit-devices #1552: ESP8266 LCtech relay x2 impossible to find correct JSON

todely, 2021-09-07:
Hello,

I have something like this:
[image]

I didn't succes to find the right JSON configuration (try all GPIO) but i think it's not possible for this x2 relay because it's serial command right ?

Thanks for your help

tonysprenk, 2021-09-07:
Yes. It uses a secondary mcu. It is compatible but you have to know the commands.

todely, 2021-09-07:
Hello
I don’t know how to find command, hope someone have same esp + relay and have found solution…

novellaip, 2021-09-08:
[image]
[image]

Worth a try.

todely, 2021-09-08:
that's exactly my module (LCtech) but how add hex command in json ?

novellaip, 2021-09-08:
Can refer to UART Configuration and UART Action in Wiki section. Since the spec. did not list the baud rate, you may need to try all of them. Typically 9600 n 8 1.

todely, 2021-09-08:
I have try this json:
{"c":{"l":13,"b":[{"g":0,"t":5}],"r":[{"n":0,"s":115200,"p":0,"b":0}]},"a":[{"t":1,"0":{"u":[{"n":0,"v":"a00101a2"}]},"1":{"u":[{"n":0,"v":"a00100a1"},]}}]}

but can't find my device with homekit ? does a wrong json avoid homekit to find device?

todely, 2021-09-08:
can't find right json for second relay...

closed 2021-09-08 by todely

novellaip, 2021-09-08:
{"c":{"l":13,"b":[{"g":0,"t":5}],"r":[{"n":0, "s":115200, "p":0, "b":0}]},"a": [{"t": 1,"0":{"u":[{"n":0,"v":"a00100a1"}]},"1":{"u":[{"n":0,"v":"a00101a2"}]}}, {"t": 1,"0":{"u":[{"n":0,"v":"a00200a2"}]},"1":{"u":[{"n":0,"v":"a00201a3"}]}}]}

You need to add one more accessory(Switch) in array “a”. Using JSON verification can help you understand.

todely, 2021-09-08:
yep i have use JSON verificator and find my mistake

ksn135, 2022-05-11:
JSON for garage doors:
`{
  "c":{"o":0,"l":13,"r":[{"n":0, "s":115200, "p":0, "b":0}]},
  "a":[{
    "t":40,
    "d":18,
    "0":{"u":[{"n":0,"v":"a00101a2","d":1000},{"n":0,"v":"a00100a1","d":1000}]},
    "1":{"a":0},
    "2":{"a":0},
    "3":{"a":0},
    "10":{"a":0},
    "12":{"a":0},
    "13":{"a":0}
  },{
    "t":40,
    "d":18,
    "0":{"u":[{"n":0,"v":"a00201a3","d":1000},{"n":0,"v":"a00200a2","d":1000}]},
    "1":{"a":0},
    "2":{"a":0},
    "3":{"a":0},
    "10":{"a":0},
    "12":{"a":0},
    "13":{"a":0}
  }]
}
`