RavenSystem/esp-homekit-devices #1553: Home App status not update immediately

novellaip, 2021-09-07:
**Detailed steps to reproduce it**
For the Status input "g[n]" and the Service Notification Action "m", Home App will not get the updated status immediately until the App was reopened again.

**JSON String**

**Used device**
Sonoff
**HAA version**
7.1.2
**iOS/iPadOS version**

**Logs**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)

**Screen-shots or videos showing issue**

RavenSystem, 2021-09-12:
JSON string is needed.

https://github.com/RavenSystem/esp-homekit-devices/wiki#issues

novellaip, 2021-09-14:
{"c":{"l":13,"b":[{"g":0,"t":5}]},
  "a":[{"0":{"r":[{"g":12}],"m": [[2, 2]]},"1":{"r":[{"g":12,"v":1}],"m": [[2, 3]]},"b":[{"g":0}],"s":0},
         {"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}],"s":0},
         {"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}],"s":0},
         {"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"g0": [{ "g": 10, "t": 1 }],"g1": [{ "g": 10, "t": 0 }],"b":[{"g":14}],"s":0}]
}

I made the test with my Home App and Sonoff 4CH Pro.
In Home App, when I press button 1, it supposed button 2 would lid but it didn't.
With Sonoff I press and hold hard button 3, it supposed Home App button 4 would lid but it didn't.
I need to re-open the Home App to see the correct result. (Buttons lid)

RavenSystem, 2021-09-14:
Thanks. It will be fixed in next HAA version.

closed 2021-09-14 by RavenSystem