RavenSystem/esp-homekit-devices #1560: No response after updating to v7.2.1

jRaufe, 2021-09-12:
Hi, great job!
After updating to v 7.2.1, all my accessories / services work perfect unless one. It is an Irrigation System with 7 services in sequence and two switches, one for enable/disable the watering and the second one is a classic dummy switch for updating. It runs in a nodemcu with this Json:

{"c":{"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":5,"v":1}],"m":[[2,1]]},"1":{"r":[{"g":5}]},"t":20,"w":1,"d":600, "es": [{"0":{"r":[{"g":4,"v":1}],"m":[[3,1]]},"1":{"r":[{"g":4}]},"t":20,"w":1,"d":600},{"0":{"r":[{"g":3,"v":1}],"m":[[4,1]]},"1":{"r":[{"g":3}]},"t":20,"w":1,"d":600},{"0":{"r":[{"g":14,"v":1}],"m":[[5,1]]},"1":{"r":[{"g":14}]},"t":20,"w":1,"d":600},{"0":{"r":[{"g":12,"v":1}],"m":[[6,1]]},"1":{"r":[{"g":12}]},"t":20,"w":1,"d":600},{"0":{"r":[{"g":13,"v":1}],"m":[[7,1]]},"1":{"r":[{"g":13}]},"t":20,"w":1,"d":600},{"0":{"r":[{"g":15,"v":1}]},"1":{"r":[{"g":15}]},"t":20,"w":1,"d":1200}]},{"t":2} ,{"1":{"s":[{"a":1}]}}]}

I have downgrade to 6.9.0 and it works perfectly. 

Second and third try to 7.2.1 with the same result: very long time in “updating” and finally “no response”.

I have also try introducing “cd”:2 for every service, but without success. 
I have also tried deleting the last dummy switch but same result

Is it an issue, or there is a way to improve the json?

RavenSystem, 2021-09-12:
It is a bug. It will be fixed soon, in v7.3.0.

jRaufe, 2021-09-12:
👍

closed 2021-09-12 by jRaufe

RavenSystem, 2021-09-12:
v7.3.0 has been released just now.