RavenSystem/esp-homekit-devices #1604: ESP01 starts into 'frc2_timer_ta'

tgoschuetz, 2021-11-01:
Using an ESP-01S with a relay, trying to create a switch for HomeKit.

Loading initial firmware to ESP, connecting to config mode and  setting WiFi, AutoUpdate (does not make a difference) and JSON (example or alternatively my own JSON, makes no difference). 
Afterwards, ESP restarts, makes two downloads and then in the final reset runs into:

'...HAAMAIN v8.1.0 installed

Restarting...
rm match
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeff78, prio:14, stack:512
pm_task_hdl : 3ffef738, prio:1, stack:176
frc2_timer_tar '

**(See Screenlog.0.txt for details up until this point)**
[screenlog.0.txt](https://github.com/RavenSystem/esp-homekit-devices/files/7453245/screenlog.0.txt)

Several resets then seem to make the ESP run again, but it seems to be incomplete:

'phy ver: 273, pp ver: 8.3

sleep disable

Home Accessory Architect v8.1.0
(c) 2018-2021 Jos√© Antonio Jim√©nez Campos

SETUP MODE
Wifi config init
mode : sta(44:17:93:0d:a1:29)
add if0
Saved BSSID: 00f76fd04d76
Wifi Mode: Normal
HAA Setup
Set hostname
Enabling auto reboot
mode : sta(44:17:93:0d:a1:29) + softAP(46:17:93:0d:a1:29)
add if1
bcn 100
Function called without core lock
Start AP SSID=HAA-0DA129
bcn 0
del if1
add if1
bcn 100
Start DHCP server
Start HTTP server
scandone
add 0
aid 5
cnt 

connected with TGWLAN, channel 6
dhcp client start...
ip:192.168.0.111,mask:255.255.255.0,gw:192.168.0.1
bcn 0
del if1
mode : sta(44:17:93:0d:a1:29)'

I can access the ESP at 192.168.0.111:4567, but no changes on this page are accepted.
Connecting the ESP to HomeKit times out after a while.

RavenSystem, 2021-11-01:
I don't see anything wrong. If you want to get logs in normal mode, add `"o":1` to JSON config section.

https://github.com/RavenSystem/esp-homekit-devices/wiki/General-Configuration#Log-output

tgoschuetz, 2021-11-02:
So maybe it's an error in the JSON although there is no error mentioned in the log
[image]
?

hennessyevan, 2021-11-05:
I had the exact same problem... not sure what it is. I'll let you know if I find a solution.

#1477

tgoschuetz, 2021-11-14:
It happens as soon as you add **any** config JSON. If you set the configuration with everything except a JSON, it boots normally afterwards. If you then enter a JSON (or right at the first configuration), it runs into this:

```
* HAAMAIN v9.1.0 installed

*** REBOOTING

rm match
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeff38, prio:14, stack:512
pm_task_hdl : 3ffef6f8, prio:1, stack:176
**frc2_timer_ta**
```

I have already tried with four different ESP01 and ESP-12 and it is sufficient to use the very basic example JSON from the wiki, it does not need to be a very complex JSON string:

`{"a":[{"t":1},{"t":2}]}`

tgoschuetz, 2021-11-14:
OK, got it using the log in normal mode.

closed 2021-11-14 by tgoschuetz

hennessyevan, 2021-11-14:
@tgoschuetz You got it? What did you need to do to get it to work?

tgoschuetz, 2021-11-15:
Hi Evan,

Without the extensive logging, it seems that the device stopped, but fact is, it is running and if the JSON is correct, you can add it to HomeKit as well.

I have configured only WiFI and OTA-Updates in the first step and then, after the updates, added just the example JSON with a „c“-section, usingthe „o“-Option Raven mentioned. This makes the ESP log more to the serial port which gives you more transparency.

In the last step, I have added the final JSON and then added it to HomeKit. (Just do some resets, so the device goes back to SETUP Mode, which you can see when monitoring the serial port.

Tom