RavenSystem/esp-homekit-devices #1607: After Flash only works for less than 1 mins

OwenL1, 2021-11-03:
Hi José, thank you for this cool project. 

I got some problem after flashing. I downloaded the 9.0.2 haamain.bin file and flash it together with the robot.bin and blank_config.bin with the commend: 

esptool.py -p /dev/cu.usbserial-110 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 haamain.bin 

After setup everything in HomeKit, the device only works for less than 1 mins then it goes to no responds. the JSON I used is shown below, I imported the JSON as one line, the following is for easy to read. Does anyone know what's going on? Thank you in advance.  
  
{
  "c": { "l": 13, "b": [{"g": 0,"t": 5}]},
  "a": [{
      "t": 1,
      "s": 5,
      "0": {"r": [{"g": 12}]},
      "1": {"r": [{"g": 12,"v": 1}]},
      "b": [{"g": 0},{"g": 4},{"g": 4,"t": 0}]
      }]
}

xrust83, 2021-11-03:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation

Didn't we switch to a single firmware file a long time ago? I think by doing this you yourself break the device by reflashing on top of the 0x1000 and 0x2000 areas

buarle, 2021-11-03:
not sure if this is related but my Gosund SP111 is unreachable after updating from 8.1 to 9.0.2...

xrust83, 2021-11-03:
I had a crash in the first minutes of the firmware release, the device was rebooted and it remained at 8.1.0, I calmed down and updated in a day. Updating to Griffron takes longer than usual, according to the logs there is a feeling that part of the data is being downloaded, the second part is being downloaded, and later the third. One of the sonoff basic hung, but after several reloads, the power was renewed by itself without opening

OwenL1, 2021-11-03:
Thank you for the prompt reply. 

I am doing a project on the local network, the workshop do not have internet for OTA, I need to install the haamain.bin to the sonoff device directly. I used the cat command to connect robot.bin and blank_config.bin to haamain.bin, I still have the same problem (device only worked for less than 1 min). Do you know how do I flash the haamain.bin to the device? 

Thanks

xrust83, 2021-11-03:
I've never done this, but I'm guessing the following.  if you run from your server, then this is just a substitution.  I think you still have to flash on fullhaaboot.bin, but when you first connect the device in the custom server window, register the parameters of your server on which you put the haamain.bin files and others in the description on the link above

Now you are trying to flash in a way that was previously called without OTA.  using files for the local server.  I think this is the problem

OwenL1, 2021-11-03:
@ xrust83  Thank you for your response. I hope someone can tell me a way to flash the haamain.bin to the device directly. This will be much easier for me.

buarle, 2021-11-04:
@xrust83: The update of the Gosund SP111 is now complete. it took more then 12 hours till it was reachable again. weird but glda that it needs no reflashing ;-)

RavenSystem, 2021-11-05:
https://github.com/RavenSystem/esp-homekit-devices/discussions/1608

closed 2021-11-05 by RavenSystem