RavenSystem/esp-homekit-devices #1609: Could not enable suggested automation “When I Leave Home, Turn Off Thermostat”

tokamac, 2021-11-04:
**Detailed steps to reproduce it**
I have two Sonoff TH16, each with its AM2301 temp+humidity sensor, connected to an electric heater and configured with HAA as "HeaterCooler with Humidity Sensor" with name "Thermostat" in two different rooms. All works fine including hourly programming, except I cannot activate the automation based on Location Services **“When I Leave Home, Turn Off Thermostat”** with an error displayed in Home.app — Everything is up-to-date from iOS to HAA. The title of this bug report is the error message displayed, that you can also see below in the attached screenshot.

**JSON String**
`{"c":{"l":13,"b":[{"g":0,"t":5}],"n":"thermostat"},"a":[{"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"5":{"a":0},"t":25,"w":1,"s":5,"g":14,"n":2,"z":0}]}`

**Used device**
iPhone X

**HAA version**
9.0.3 Griffon

**iOS/iPadOS version**
iOS 15.1

**Screen-shot showing issue**
[image]

tokamac, 2021-11-04:
Just updated through OTA all my compatible accessories to HAA 9.0.4, but issue is still the same.

However, this may be a HomeKit issue and not an HAA issue (?) because if I set up a similar automation manually, it seems to work as expected (see screenshot below). What does not work is Apple's preconfigured automation with the virtual green switch in Home.app

As a conclusion, this issue is a minor one as there is an easy (manual) workaround.

[image]

Soldiiier, 2021-11-08:
I get a similar issue with an offical supported Hk thermostat device, when enabling suggested automations. I don't think this is attributed to HAA

RavenSystem, 2021-11-10:
This issue is related to HomeKit, not HAA. Feel free to create a Discussion to get help. Thanks!

closed 2021-11-10 by RavenSystem