RavenSystem/esp-homekit-devices #1612: How to flash haamain.bin?

OwenL1, 2021-11-06:
Hi José and everyone, 

Is that possible to flash the haamain.bin directly into the esp8266? I don't have internet to use the OTA at the workshop. I am really new to programming and I could not figure out how to setup the local server. I have a usb to serial adapter I can use. Can anyone please give me some guidance? 

Here is the problem I have: 
I download the haamain.bin, I use the cat command to combine the rboot.bin and blank_config.bin with haamain.bin. Then flashed the final .bin file into the esp8266, I can setup the switch in HomeKit, it works, but only for less than 1mins. Then the device went into a no respond state.      

I really want to know what's going on here? Is there something I missed? 

Thank you very much

gerarddvb, 2021-11-07:
If you can add switch in HomeKit it means that the flashing process is done correctly. There is probably an issue with your wifi router. My router had an issue with a power saving state witch dropped all HAA devices after a couple of minutes

OwenL1, 2021-11-08:
@gerarddvb thank you for the response, it was not the issue with the router. I have a switch on the older version HAA with this router and it is working perfectly.

RavenSystem, 2021-11-10:
Sorry, you must flash fullhaaboot.bin file and then install other files using OTA. Offline installations support was dropped time ago to get more flash storage and add more features.

closed 2021-11-10 by RavenSystem