RavenSystem/esp-homekit-devices #1613: IR Not seen in HAA

sgsteyn, 2021-11-06:
All of my HAA devices are visible and manageable in the HAA app except my IR hub which is completely missing but working flawlessly in the Home app.

I see there were some post back in 2019 that IR is not supported in HAA, has anything changed?

`{"c":{"n":"living-ir","t":14,"p":"J(J(AcAcAcDFAc"},"a":[{"t":60,"xa":0,"i":[{"n":"CCTV","0":{"i":[{"r":3,"c":"E2AsAkAmAkAkB*AlB*AlC0AbAkAkB*AlAlApB(AlAkAkB*ApB*AlAgAkAlAlAkw0E&AgAkApAgApB(ApB&AlB*AlAlApB(AlAkAlB-AgAlAlB:AlB(ApAgApAgAkAkv-E(AgAgAkAlApB&AlB-AlB*AlAgApB,AkAgApB(ApAhAoB*AlB*AlAgApAgApAg"}]}}],"0":{"i":[{"r":3,"w":"E3AqCBAcB:AmB*AlB*AkAgApB(ApAgApB*AlAgAoAhAoAgApAg7JE(AhB(ApB_AgB(ApB&ApAgApB*AkAhAoB,AkAgApAgApAgAoAg7VE!AjB,AkB(ApB,AkB,AkAhApB(ApAgApB*AlAgApAgApAgApAg"}]},"1":{"i":[{"r":3,"w":"E2AxAgAqB*AlB-AgB(AkAlApB,AlAgApB(AkAlApAgApAgApAh8rE@AmAgApB,AfB,ApB(AqAgAkB_AkAgApB*AkAgArAgAmAhApAg82E@AlAgApB,ApB&AkB,AlAgAoB,ApAbApB*AlAgApAgApAgAuAb"}]}},{"i":0.5,"1":{"s":[{"a":1}]}}]}`

Blitzwolf RC1
HAA version 2.0.0
IOS version 15.1

tonysprenk, 2021-11-06:
It has an older version of haa which is not supported by the haa manager app. Update to the latest version and it will show up.

sgsteyn, 2021-11-07:
Also thought so but update yesterday from 8.0.0 to 9.0.4 but didn't make a difference l
[image]

xrust83, 2021-11-07:
In fact, this is a TV service that should not be supported by a HAA manager.

[image]

sgsteyn, 2021-11-08:
Thank you, remembered it about that now.

closed 2021-11-08 by sgsteyn