RavenSystem/esp-homekit-devices #1615: Hola!!!! no puedo controlar automatización con temperatura del th16 con sensor si7021

Danilo0609, 2021-11-08:
Configuré mi sonoff th 16 con sensor si 7120 con HAA Griffon. funciona perfecto todo excepto que la automatización basada en temperatura no funciona. funciona perfectamente la lectura de temperatura y humedad pero si deseo que se apague o encienda a una temperatura indicada no lo puedo hacer. no esta habilitada esa función? esta bien el json que use?? agradeceré mucha la ayuda!!!!

**JSON String**

{"c":{"l":13,"b":[{"g":0,"t":5}],"o":1},"a":[{"s":5,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":24,"g":14,"n":4,"j":10}]}
**Used device**

**HAA version**

Griffon 9.0.4 

**iOS/iPadOS version**

OS 14.7.1 iPhone x

**Logs**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)

**Screen-shots or videos showing issue**

RavenSystem, 2021-11-10:
https://github.com/RavenSystem/esp-homekit-devices/discussions/1616

closed 2021-11-10 by RavenSystem

Danilo0609, 2021-11-10:
Muchísimas gracias por la respuesta. He podido solucionar parte del problema con estos json.

EL json que utilice es

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"5":{"a":0},"t":21,"s":5,"g":14,"b":[],"n":4,"j":10,"w":1,"m":10,"x":40},{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":1,"b":[{"g":0,"t":1}],"s":0},{"t":24,"b":[],"n":4,"g":14,"j":10}]}

Pero no este json no puedo controlar el dispositivo con respecto a la humedad y no puedo establecer un rango de grados entre el encendido y el apagado del th16 como con el firmware original de sonoff. Se podrá configurar un json completo?. 

Nueva consulta. Como reseteo el wifi de un Device ya instalado para cambiarlo? Con el boton fisico no me lo permitió. Lo presione durante 8 segundos y no me deja conectarme  A LA RED HAA. Da error. No se como resetearlo. 

Muchísimas gracias. Excelente post!!!!!!!