RavenSystem/esp-homekit-devices #1630: ESP8266EX problem.

nickforev, 2021-11-16:
Hi. I found bug in release 9.1.0. Sonoff TH16 ver 2.1 after flash dont load. Chip ESP8266EX problem. I downgrade to release 8.0.0 and all works fine.
[image]

nickforev, 2021-11-16:
After update work normal. 
[image]

VaBanck, 2021-11-16:
А в чём проблема? Я так и не понял

nickforev, 2021-11-17:
> А в чём проблема? Я так и не понял
Проблема в том что на последней прошивке esp8266ex не запускается. Только если зашить 8.0.0, а потом обновить работает.
Проверено на нескольких ТН16, все одинаково.
Это отчет про случайно найденый баг.

nickforev, 2021-11-22:
HAA 9.2.0 same problem!

RavenSystem, 2021-11-22:
I have tested it, and works as expected. About 10 seconds after flash, HAA wifi hotspot is reachable.

RavenSystem, 2021-11-23:
Could you try this, please? (Unzip it first)
[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7586979/fullhaaboot.bin.zip)

RavenSystem, 2021-11-23:
Try this, please:

[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7587768/fullhaaboot.bin.zip)

RavenSystem, 2021-11-23:
https://github.com/RavenSystem/esp-homekit-devices/issues/1637

closed 2021-11-23 by RavenSystem