RavenSystem/esp-homekit-devices #1637: Wifi in Shelly RGBW2 is not running

pklejnowski, 2021-11-22:
**Detailed steps to reproduce it**
I'm trying to flash device Shelly RGBW2 with following command:
`esptool.py -p COM3 erase_flash`
`esptool.py -p COM3 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 fullhaaboot.bin`
It seems that deviced is flashed successfully, although wifi network HAA-XXXXX is not starting.
Does anyone have idea what i've done incorrectly? It's my first Shelly RGBW2 device, before that I was flashing in the same way Sonoff switches - and I hadn't any problem.

**Used device**
Shelly RGBW2

**HAA version**
Latest (tried on a few previous too)

**Screen-shots or videos showing issue**
[image]

RavenSystem, 2021-11-22:
I have tested it, and works as expected. About 10 seconds after flash, HAA wifi hotspot is reachable.

pklejnowski, 2021-11-22:
Command that I used for flashing is correct?
So might it be the problem with device? Can I do anything to verify it somehow?

RavenSystem, 2021-11-22:
Yes, command is right. You can try to read serial logs using PuTTY.

pklejnowski, 2021-11-22:
Do you have any link with tutorial how can I do this?

RavenSystem, 2021-11-22:
For example: https://www.eye4software.com/hydromagic/documentation/articles-and-howtos/serial-port-logging/

Speed must be 115200

pklejnowski, 2021-11-22:
Should I do this whilst flashing? If start putty, then I cannot flash because of Permission denied error.

```
  File "C:\Users\divao\AppData\Local\Programs\Python\Python38-32\Scripts\esptool.py-script.py", line 11, in <module>
    load_entry_point('esptool==2.8', 'console_scripts', 'esptool.py')()
  File "c:\users\divao\appdata\local\programs\python\python38-32\lib\site-packages\esptool.py", line 3194, in _main
    main()
  File "c:\users\divao\appdata\local\programs\python\python38-32\lib\site-packages\esptool.py", line 2883, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "c:\users\divao\appdata\local\programs\python\python38-32\lib\site-packages\esptool.py", line 273, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "c:\users\divao\appdata\local\programs\python\python38-32\lib\site-packages\esptool.py", line 237, in __init__
    self._port = serial.serial_for_url(port)
  File "c:\users\divao\appdata\local\programs\python\python38-32\lib\site-packages\serial\__init__.py", line 88, in serial_for_url
    instance.open()
  File "c:\users\divao\appdata\local\programs\python\python38-32\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, 'Permission denied.', None, 5)
```

RavenSystem, 2021-11-22:
No, after flash.

pklejnowski, 2021-11-22:
Logs are empty. 
I was thinking then putty is run too late so I was even trying to put programmator first. 
Then run putty.
Then plug cables to Shelly device, but it throw an error: 
[image]

RavenSystem, 2021-11-22:
Once flash is finished, disconnect Shelly, prepare PuTTY, connect Shelly, and run PuTTY.

pklejnowski, 2021-11-22:
Log file isn't created.

pklejnowski, 2021-11-22:
If I plug off cable (for example VCC cable), then log is created but is empty.

pklejnowski, 2021-11-23:
Shelly's seller gave me his bin (it includes JSON) - no idea but it works with his bin. 
How it is even possible? :o Maybe because it's not the latest version? And in the newest there is a bug?

[HKshelly_RGBW.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7586890/HKshelly_RGBW.zip)

RavenSystem, 2021-11-23:
Bug is in latest versions, but only happens with some devices.

pklejnowski, 2021-11-23:
With that shelly rgbw2 too?

RavenSystem, 2021-11-23:
Yes. Bug is caused when you flash it in a new device that never had HAA installed.

RavenSystem, 2021-11-23:
Have you any other device to do some tests?

pklejnowski, 2021-11-23:
Other than Shelly RGBW2? Yes I have sonoff touch, but it seems working properly.

RavenSystem, 2021-11-23:
Other device that fails. Have you got any?

pklejnowski, 2021-11-23:
No, i'll let you know if I will find any yet. Because I'm going to flash other devices for my home soon.
Btw. do you know when can we expect patched version? With my current HAA version (4.9) I won't be able to use it with HAA Manager application that I bought from appstore.

RavenSystem, 2021-11-23:
If/When I find a device with the bug, very soon (a few hours).

RavenSystem, 2021-11-23:
BTW, you can reflash your RGBW2 with last fullhaaboot.bin. Now, it will work.

pklejnowski, 2021-11-23:
> BTW, you can reflash your RGBW2 with last fullhaaboot.bin. Now, it will work.

But should I omit the step: `esptool.py -p COM3 erase_flash`?

RavenSystem, 2021-11-23:
Better to erase_flash.

pklejnowski, 2021-11-23:
Still the same problem with the newest version. Wifi is not showin up.

pklejnowski, 2021-11-23:
I also noted that flashing with the newest version takes ~30seconds, and with bin that I uploaded takes much longer - 74 seconds.

RavenSystem, 2021-11-23:
Flashing time is normal. Fullhaaboot is only a part of the firmware. Let me send you other fullhaaboot to test, please.

pklejnowski, 2021-11-23:
Ok

RavenSystem, 2021-11-23:
[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7587478/fullhaaboot.bin.zip)

pklejnowski, 2021-11-23:
I've checked your bin. Still not working.

RavenSystem, 2021-11-23:
Other test.

[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7587579/fullhaaboot.bin.zip)

RavenSystem, 2021-11-23:
`{"c":{"l":2,"b":[{"g":13,"t":5}],"n":"hk_rgbw"},"a":[{"t":30,"g":[15,12,14,4],"b":[{"g":5,"p":0},{"g":5,"t":0},{"g":13}]}]}`

RavenSystem, 2021-11-23:
Try this one, please:

[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7587759/fullhaaboot.bin.zip)

xrust83, 2021-11-23:
let me interfere. When copying the json code, I developed a situation when, due to the keyboard layout and language, for example, the quotes did not match the encoding. and if you look closely, you can see that they even look differently. accordingly, such a code was not saved and haa the device was rebooted and re-entered the configuration mode. To avoid this, enter json into the system, or, before inserting it into the device, first insert it into the address, into any window where styles and text encodings are killed. It helped me once.

I hope the translator will translate my thought correctly

pklejnowski, 2021-11-23:
> `{"c":{"l":2,"b":[{"g":13,"t":5}],"n":"hk_rgbw"},"a":[{"t":30,"g":[15,12,14,4],"b":[{"g":5,"p":0},{"g":5,"t":0},{"g":13}]}]}`

I was trying to use that code but with old version of HAA, and I couldn't change colors, it was behaving like a normal light (only brightness could be changed).

> Try this one, please:
> 
> [fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7587759/fullhaaboot.bin.zip)

Same, doesn't work.

RavenSystem, 2021-11-23:
And this?

[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7588012/fullhaaboot.bin.zip)

pklejnowski, 2021-11-23:
Nope.

RavenSystem, 2021-11-23:
Other chance:

[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7588232/fullhaaboot.bin.zip)

pklejnowski, 2021-11-23:
Still doesn't work.

RavenSystem, 2021-11-23:
Other:
[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7588274/fullhaaboot.bin.zip)

pklejnowski, 2021-11-23:
No. I admire your spirit :)

RavenSystem, 2021-11-23:
I don't understand what is happening. Can you try this old version?

https://github.com/RavenSystem/haa/releases/download/8.0.0/fullhaaboot.bin

pklejnowski, 2021-11-23:
Yes, that version is working.

RavenSystem, 2021-11-23:
And this?

https://github.com/RavenSystem/haa/releases/download/9.0.4/fullhaaboot.bin

pklejnowski, 2021-11-23:
It works fine

RavenSystem, 2021-11-23:
Again: 

https://github.com/RavenSystem/haa/releases/download/9.1.0/fullhaaboot.bin

pklejnowski, 2021-11-23:
No.

RavenSystem, 2021-11-23:
Ok, we are close. Issue was introduced in 9.1.0.

RavenSystem, 2021-11-23:
Other more:
[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7588984/fullhaaboot.bin.zip)

pklejnowski, 2021-11-23:
Still no.

RavenSystem, 2021-11-23:
Try to capture logs using PuTTY. It must show "Hello" each second.

[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7589042/fullhaaboot.bin.zip)

pklejnowski, 2021-11-23:
[putty.log](https://github.com/RavenSystem/esp-homekit-devices/files/7589248/putty.log)

RavenSystem, 2021-11-23:
Very useful!

RavenSystem, 2021-11-23:
Try this, and capture logs again if doesn't work.
[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7589305/fullhaaboot.bin.zip)

pklejnowski, 2021-11-23:
I will try to add it to HomeKit, just wait for it :D

RavenSystem, 2021-11-23:
You can use that binary to install your Shelly RGBW2. I will release a new version soon.

RavenSystem, 2021-11-23:
And, of course, many thanks for your help!! Feel free to join to official Discord chat. I am there, with other users.

pklejnowski, 2021-11-23:
> And, of course, many thanks for your help!! Feel free to join to official Discord chat. I am there, with other users.

Definitely I will! Especially that I'm nearly finish home renovation, and I will have a lot of devices on HAA :D 

Btw. on blackfriday are you planning some promos for HAA Manager tool from appstore? Not sure if I can buy it now or maybe I should hold a few days :D

RavenSystem, 2021-11-23:
Better join to Discord chat to talk about other matters than this issue. Use @RavenSystem to mention me.

pklejnowski, 2021-11-23:
Thank you for all your work :) I think you can close this issue, just let us know when the fixed version will be released :D

RavenSystem, 2021-11-23:
Are you in Discord?

[2 more replies]

closed 2021-11-23 by RavenSystem