RavenSystem/esp-homekit-devices #1642: Sonoff SV after HAA flash not showing AP WIFI

Herwarth2019, 2021-11-22:
Hello Mr. José Antonio Jiménez Campos,

firstly thank you for your great work on this DIY Homekit system - the HAA. I used your software till first RavenCore and was 100% satisfied after that I migrated to the HAA and was much happier because of JSON possibilities... Yesterday I came back to your software to update my Sonoff SV devices but with that I stucked in the 3. part of your installation guidline - when I flash the Sonoff SV (everything went well - see screenshot) the device wont show the Wifi AP so Iam not able to connect it. Iam able to repeat the 1.-3. step (everything actual) but still it stucks in "not showing wifi AP - cant see "HAA...". I tried other devices (Sonoff Basic, RF,..) but just the SV do this issue :-/... Iam getting out of my mind because tried around 80-100 flashes/per one device and sometimes it works like 20-30 flashes/1 working SV but mostly it gets stuck without the wifi AP... Do you know whats happening? note tried around 5 new Sonoff SV everything do the same...

Thanks for any advice,
Best regards
David

[image]

Herwarth2019, 2021-11-22:
update: tried for the first time to install latest Tasmota - just changed the .bin file and flashed. Worked 100% and after reboot in around 2s shows wifi AP tasmota. Then I put the fullhaaboot and it did the same issue again :-/...So I checked that the device and connection work there must be something else...

RavenSystem, 2021-11-22:
Could you try this, please? (Unzip it first)
[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7584872/fullhaaboot.bin.zip)

RavenSystem, 2021-11-23:
Try this, please:

[fullhaaboot.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7587763/fullhaaboot.bin.zip)

xrust83, 2021-11-23:
let me interfere.  When copying the json code, I developed a situation when, due to the keyboard layout and language, for example, the quotes did not match the encoding.  and if you look closely, you can see that they even look differently.  accordingly, such a code was not saved and haa the device was rebooted and re-entered the configuration mode.  To avoid this, enter json into the system, or, before inserting it into the device, first insert it into the address, into any window where styles and text encodings are killed.  It helped me once.

RavenSystem, 2021-11-23:
https://github.com/RavenSystem/esp-homekit-devices/issues/1637

closed 2021-11-23 by RavenSystem

Herwarth2019, 2021-11-24:
Thank you! It works! :-) Have a great day. DH