RavenSystem/esp-homekit-devices #1661: Update all HAA devices brick my entire home (9.1.1 -> 9.5)

Alexandre-Mainardi, 2021-12-04:
**Detailed steps to reproduce it**
My devices was on 9.1.1 (21 Sonoff mini + Sonoff Touch)
I just open HAA manager (V2.0.0) and click "Update all HAA device" as I always do, to update to 9.5
My entire home become off-line (Even the touch operation on Sonoff touch units) I wait for 24h and nothing happens

I Check my router and my devices aren't connected to my wifi network.

I already do a power reset on my home and nothing.

**Used device**
Sonoff Touch, Sonoff Mini, Electrodragon ESP Relay board

**HAA version**
9.1.1 to 9.5

**iOS/iPadOS version**
iPhone 11 / IOS 15.0.1

SyBeer, 2021-12-04:
I have the same problem with update from 8.4 to 9.5 on Sonoff SV. Update hangs on last line (all logs in attached file screenlog_hang.txt) 

```
* HAAMAIN v9.5.0 installed
*** REBOOTING
rm match
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeff38, prio:14, stack:512
pm_task_hdl : 3ffef648, prio:1, stack:176
frc2_timer_ta
```

I try few times. Once update finished but there is someting strange, and device didn't respond

```
* HAAMAIN v9.5.0 installed
*** REBOOTING
rm match
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeff38, prio:14, stack:512
pm_task_hdl : 3ffef648, prio:1, stack:176
frc2_timer_ta
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeff38, prio:14, stack:512
pm_task_hdl : 3ffef648, prio:1, stack:176
frc2_timer_taĪēēŃĀdisable
mode : sta(8c:aa:b5:65:c5:1a)
add if0

[screenlog_hang.txt](https://github.com/RavenSystem/esp-homekit-devices/files/7654415/screenlog_hang.txt)
Home Accessory Architect v9.5.0
(c) 2018-2021 Jos√© Antonio Jim√©nez Campos

SETUP MODE
Wifi config init
Saved BSSID: 382c4a68f238
Wifi Mode: Normal
HAA Setup
Set hostname
! Invalid JSON
mode : sta(8c:aa:b5:65:c5:1a) + softAP(8e:aa:b5:65:c5:1a)
add if1
bcn 100
Function called without core lock
Start AP SSID=HAA-65C51A
bcn 0
del if1
add if1
bcn 100
Start DHCP server
Start HTTP server
scandone
add 0
aid 9
cnt 

connected with AS_WLAN, channel 8
dhcp client start...
ip:192.168.1.18,mask:255.255.255.0,gw:192.168.1.1
bcn 0
del if1
mode : sta(8c:aa:b5:65:c5:1a)
```

look at the code **frc2_timer_taĪēēŃĀdisable**

Daveblanche, 2021-12-04:
Im not Good with these logs, but if you look at the last few lines it seems to say that JSON is invalid. Also that ap started…but then gives WLAN ip. Have you searched for the Ssid shown and tried to connect to it?

RavenSystem, 2021-12-04:
@Alexandre-Mainardi Logs are needed to know what is happening during update.

RavenSystem, 2021-12-04:
@SyBeer I don't see anything wrong. Your log is right. If you want to see logs in HAA normal mode, you must enable logs in JSON: https://github.com/RavenSystem/esp-homekit-devices/wiki/General-Configuration#Log-output

KirillMT, 2021-12-04:
I have a new ESP12F. After loading fullhaaboot.bin ver 9.5.0 esp in loop already 12 hours. Log below. 
 
 
Dec 05 090411 HAA Log.log

RavenSystem, 2021-12-05:
@KirillMT Could you try this fullhaaboot beta and check if it downloads first part, please?

[fullhaaboot_b.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/7655719/fullhaaboot_b.bin.zip)

SyBeer, 2021-12-05:
> @SyBeer I don't see anything wrong. Your log is right. If you want to see logs in HAA normal mode, you must enable logs in JSON: https://github.com/RavenSystem/esp-homekit-devices/wiki/General-Configuration#Log-output
> 
> Edit: @Daveblanche is right, log shows Invalid JSON.

Yes, I this case JSON was empty. But I can't run SETUP mode to change that. 

At the end I flashed my devices from my local webserver using HAA 9.1.1. 
And that works for me.

RavenSystem, 2021-12-05:
> Yes, I this case JSON was empty. But I can't run SETUP mode to change that.

Setup mode is enabled automatically when JSON is invalid, as you can see at the end of your log:

Home Accessory Architect v9.5.0
(c) 2018-2021 Jos√© Antonio Jim√©nez Campos

SETUP MODE
Wifi config init
Saved BSSID: 382c4a68f238
Wifi Mode: Normal
HAA Setup
Set hostname
! Invalid JSON
mode : sta(8c:aa:b5:65:c5:1a) + softAP(8e:aa:b5:65:c5:1a)
add if1
bcn 100
Function called without core lock
Start AP SSID=HAA-65C51A
bcn 0
del if1
add if1
bcn 100
Start DHCP server
Start HTTP server
scandone
add 0
aid 9
cnt 

connected with AS_WLAN, channel 8
dhcp client start...
ip:192.168.1.18,mask:255.255.255.0,gw:192.168.1.1
bcn 0
del if1
mode : sta(8c:aa:b5:65:c5:1a)

SyBeer, 2021-12-05:
I don't understand what's happened on 9.5, and why it works on 9.1.1. Thank you for U answers.

RavenSystem, 2021-12-05:
v9.5.1 has enough improvements to avoid SSL/TLS errors, caused by low DRAM.

closed 2021-12-05 by RavenSystem