RavenSystem/esp-homekit-devices #1671: Shelly 1L can't download HAAMAIN

i3laze, 2021-12-21:
Just flashed and configured via 192.168.4.1:4567.

Can't download HAA firmware from GitHub...  It could be temporary, or something bigger- please investigate.

```
DNS..OK Socket..OK Connect..OK SSL..OK github.com:443..OK
Write OK
GOT 3383 Bytes:
HTTP/1.1 302 Found
Server: GitHub.com
Date: Tue, 21 Dec 2021 18:19:04 GMT
Content-Type: text/html; charset=utf-8
Vary: X-PJAX, X-PJAX-Container, Accept-Encoding, Accept, X-Requested-With
permissions-policy: interest-cohort=()
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211221%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211221T181904Z&X-Amz-Expires=300&X-Amz-Signature=b4f2be9833ec7c0509ff54daf009e8985c9951ebf7f411bab1efc9eef07bb586&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
Cache-Control: no-cache
Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
X-Frame-Options: deny
X-Content-Type-Options: nosniff
X-XSS-Protection: 0
Referrer-Policy: no-referrer-when-downgrade
Expect-CT: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
Content-Security-Policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.githubapp.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com wss://alive.github.com *.actions.githubusercontent.com wss://*.actions.githubusercontent.com online.visualstudio.com/api/v1/locations raw.githubusercontent.com github-production-repository-image-32fea6.s3.amazonaws.com github-production-release-asset-2e65be.s3.amazonaws.com insights.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src render.githubusercontent.com viewscreen.githubusercontent.com notebooks.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com collector.githubapp.com github-cloud.s3.amazonaws.com secured-user-images.githubusercontent.com/ *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/
Content-Length: 654
X-GitHub-Request-Id: C7AF:13A0B:1CD8D28:1E0E02C:61C21AFB

<html><body>You are being <a href="https://objects.githubusercontent.com/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211221%2Fus-east-1%2Fs3%2Faws4_request&amp;X-Amz-Date=20211221T181904Z&amp;X-Amz-Expires=300&amp;X-Amz-Signature=b4f2be9833ec7c0509ff54daf009e8985c9951ebf7f411bab1efc9eef07bb586&amp;X-Amz-SignedHeaders=host&amp;actor_id=0&amp;key_id=0&amp;repo_id=166993980&amp;response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&amp;response-content-type=application%2Foctet-stream">redirected</a>.</body></html>

HTTP returns 302

*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211221%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211221T181904Z&X-Amz-Expires=300&X-Amz-Signature=b4f2be9833ec7c0509ff54daf009e8985c9951ebf7f411bab1efc9eef07bb586&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211221%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211221T181904Z&X-Amz-Expires=300&X-Amz-Signature=b4f2be9833ec7c0509ff54daf009e8985c9951ebf7f411bab1efc9eef07bb586&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211221%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211221T181904Z&X-Amz-Expires=300&X-Amz-Signature=b4f2be9833ec7c0509ff54daf009e8985c9951ebf7f411bab1efc9eef07bb586&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211221%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211221T181904Z&X-Amz-Expires=300&X-Amz-Signature=b4f2be9833ec7c0509ff54daf009e8985c9951ebf7f411bab1efc9eef07bb586&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211221%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211221T181904Z&X-Amz-Expires=300&X-Amz-Signature=b4f2be9833ec7c0509ff54daf009e8985c9951ebf7f411bab1efc9eef07bb586&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211221%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211221T181904Z&X-Amz-Expires=300&X-Amz-Signature=b4f2be9833ec7c0509ff54daf009e8985c9951ebf7f411bab1efc9eef07bb586&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
! Tries 2
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECTION
reconnect
scandone
Saved BSSID: e447b3fe911e
Wifi Mode: Normal

* SETUP

mode : sta(e8:db:84:a2:3f:4e) + softAP(ea:db:84:a2:3f:4e)
add if1
bcn 100
Function called without core lock
Start AP SSID=HAA-A23F4E
bcn 0
[... 493 lines cut ...]
*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211221%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211221T182322Z&X-Amz-Expires=300&X-Amz-Signature=f1ad3a4373d8ad3dac497e6f9477f47bef42302c5364602bd5e7a8c5e521017b&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attach [... cut ...] =300&X-Amz-Signature=f1ad3a4373d8ad3dac497e6f9477f47bef42302c5364602bd5e7a8c5e521017b&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
! Tries 5
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECTION

```

i3laze, 2021-12-22:
Beginning to wonder if ESP8266EX and Shelly 1L is actually supported.

I've deployed custom HTTP server and tried 9.5.0 and 7.6.0 binaries.
Here's I catched a kernel panic using 7.6.0 binaries (HAA Installer v.5.2.0)

```
rBoot Option: Big flash`
`rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffefab8, prio:14, stack:512
pm_task_hdl : 3ffef3a8, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4368, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Nov 17 2021 19:14:39
phy ver: 273, pp ver: 8.3

sleep disable

Sysparam OK

Wifi Init
mode : sta(e8:db:84:a2:3f:4e)
add if0
Wifi reset
scandone
no none found, reconnect after 1s
reconnect
scandone
Saved BSSID: e447b3fe911e
Wifi Mode: Normal

HAA INSTALLER - NORMAL MODE

scandone
add 0
aid 2
cnt

connected with Sweet Home, channel 10
dhcp client start...
ip:192.168.1.3,mask:255.255.255.0,gw:192.168.1.1

HAA Installer v5.5.2

- Server 192.168.1.11
- Port   80
- SSL    no

Current HAAMAIN installed v0.0.0

INIT

*** STARTING UPDATE PROCESS

Running HAABOOT

>>> Get sign
*** DOWNLOADING FILE
*** FORWARDING: 192.168.1.11:80/otamain.bin.sec
*** NEW CONNECTION
DNS..OK Socket..OK Connect..OK Write..OK
GOT 336 Bytes:
HTTP/1.1 200 OK
Server: MiniWeb
Cache-control: no-cache
Pragma: no-cache
Connection: keep-alive
Accept-Ranges: bytes
Last-Modified: Wed, 22 Dec 2021 22:38:30 GMT
[... 1899 lines cut ...]
Wifi Mode: Normal

HAA INSTALLER - SETUP MODE

mode : sta(e8:db:84:a2:3f:4e) + softAP(ea:db:84:a2:3f:4e)
add if1
bcn 100
Function called without core lock
Start AP SSID=HAA-A23F4E
bcn 0
del if1
add if1
bcn 100
Start DHCP server
Start HTTP server
scandone
add 0
aid 2
cnt

connected with Sweet Home, channel 10
dhcp client start...
ip:192.168.1.3,mask:255.255.255.0,gw:192.168.1.1

```

i3laze, 2021-12-25:
Out of the blue... Shelly 2.5 has the same symtoms.. SSL -308.

i3laze, 2021-12-25:
Out of the blue... Shelly 2.5 has the same symtoms.. SSL -308.
@RavenSystem , please take a look.

RavenSystem, 2021-12-25:
Your local server is incompatible, because it is an incomplete web server that lacks partial content support.

And about GitHub server issue, I can not replicate your issue. ESP2866 is sold in two chip formats, because ESP8266 itself is not a chip. Those 2 chip formats are ESP8266EX and ESP8285.

I have done some modifications in HAA OTA Installer. Download and flash `fullhaaboot.bin` from beta repo, and then configure custom server to use beta repo. Please, comment results.

https://github.com/RavenSystem/haabeta

i3laze, 2021-12-25:
> Your local server is incompatible, because it is an incomplete web server that lacks partial content support.
>
> And about GitHub server issue, I can not replicate your issue. ESP2866 is sold in two [...]

Sorry, no luck with v5.8.0 installer.. 
I'll try different ISP and local web server to eliminate these factors now.

Logs:

add 1
aid 1
station: ce:d1:94:e9:98:22 join, AID = 1
Client disconnected
Client disconnected
Start Wifi scan
Client disconnected
scandone
Start Wifi scan
Client disconnected
scandone
Start Wifi scan
scandone
Client disconnected
Stop HTTP server
Update settings

Rebooting
del if0
station: ce:d1:94:e9:98:22 leave, AID = 1
rm match
bcn 0
del if1
usl
sul 0 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffefab8, prio:14, stack:512
pm_task_hdl : 3ffef3a8, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4368, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Dec 25 2021 16:29:19
phy ver: 273, pp ver: 8.3

sleep disable
mode : sta(c4:5b:be:6a:a2:f8)
add if0
Sysparam OK

Wifi Init

HAA INSTALLER
Wifi clean
scandone
[... 256 lines cut ...]
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211225%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211225T165226Z&X-Amz-Expires=300&X-Amz-Signature=fa46cb153b5da5a360f9b76909ec7e1f8b37a145f1d4b5f629f7f34bee2fd5fa&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211225%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211225T165226Z&X-Amz-Expires=300&X-Amz-Signature=fa46cb153b5da5a360f9b76909ec7e1f8b37a145f1d4b5f629f7f34bee2fd5fa&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211225%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211225T165226Z&X-Amz-Expires=300&X-Amz-Signature=fa46cb153b5da5a360f9b76909ec7e1f8b37a145f1d4b5f629f7f34bee2fd5fa&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211225%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211225T165226Z&X-Amz-Expires=300&X-Amz-Signature=fa46cb153b5da5a360f9b76909ec7e1f8b37a145f1d4b5f629f7f34bee2fd5fa&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
*** FORWARDING: objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/31b47706-4537-444c-9e7c-ef90bb70c438?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211225%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211225T165226Z&X-Amz-Expires=300&X-Amz-Signature=fa46cb153b5da5a360f9b76909ec7e1f8b37a145f1d4b5f629f7f34bee2fd5fa&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
*** CONNECTION
DNS..OK Socket..OK Connect..OK SSL..OK objects.githubusercontent.com:443..! Connect -1
! SSL -308
! Tries 3
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECTION

i3laze, 2021-12-25:
@RavenSystem , for some reason it was GitHub..  
I've tried IIS 2019 web server and it done the job.
I guess now I should try with SSL.

Alexvrrr, 2021-12-28:
Hello .  i have the same problem on the device ESP 1

XDimik, 2022-01-03:
I have the same problem on the device Sonoff SV

RavenSystem, 2022-01-03:
Try this: https://github.com/RavenSystem/haabeta/releases/tag/9.56.1

i3laze, 2022-01-03:
Happy New Year!
Will try within few hours.

RavenSystem, 2022-01-03:
@i3laze Can you do `ping objects.githubusercontent.com` and tell me what IP address is?

RavenSystem, 2022-01-03:
Mine is `185.199.108.133`

i3laze, 2022-01-03:
[image]
Although currently I'm using a different ISP..

Got Shelly 2.5 connected, please stand by for logs.

i3laze, 2022-01-03:
And.. looks like under current ISP  OTA update cycle worked just fine!!

<details><summary>Logs</summary>
add 1
aid 1
station: f6:9e:99:2e:e0:e2 join, AID = 1
Client disconnected
Start Wifi scan
scandone
Client disconnected
! Unknown: GET /favicon.ico
Unknown
Redirecting to http://192.168.4.1:4567/settings
Client disconnected
Start Wifi scan
scandone
Client disconnected
Start Wifi scan
scandone
Client disconnected
Stop HTTP server
Update settings
Wifi clean
scandone
no none found, reconnect after 1s
reconnect

Rebooting
scandone
no none found, reconnect after 1s
del if0
station: f6:9e:99:2e:e0:e2 leave, AID = 1
rm match
bcn 0
del if1
usl
sul 0 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeff38, prio:14, stack:512
pm_task_hdl : 3ffef648, prio:1, stack:176
frc2_timer_ta
add 0
aid 13
cnt

connected with WiFi, channel 1
dhcp client start...
ip:10.0.0.77,mask:255.255.255.0,gw:10.0.0.254
bcn 0
del if1
mode : sta(c4:5b:be:6a:a2:f8)
Start Wifi scan
[... 299 lines cut ...]

load 0x40100000, len 2292, room 16
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeff38, prio:14, stack:512
pm_task_hdl : 3ffef648, prio:1, stack:176
frc2_timer_ta

</details>

Sorry, can't test your beta firmware, as here and now GitHub just works.
Anyone else, mbe?

RavenSystem, 2022-01-03:
Forget new BETA. Other user tested it and get -308 errors.

RavenSystem, 2022-01-03:
I think now it is solved. Try last BETA:

https://github.com/RavenSystem/haabeta/releases/tag/9.56.3

XDimik, 2022-01-04:
nslookup objects.githubusercontent.com
╤хЁтхЁ:  RT-AX68U-7198
Address:  192.168.1.1

Не заслуживающий доверия ответ:
╚ь :     objects.githubusercontent.com
Addresses:  185.199.108.133
          185.199.111.133
          185.199.110.133
          185.199.109.133

XDimik, 2022-01-04:
> Думаю, теперь это решено. Попробуйте последнюю БЕТА-версию:
> 
> https://github.com/RavenSystem/haabeta/releases/tag/9.56.3

It didn't work. Tried to update my old devices over the air from version 6.0.7 and 7.3.1. Also failed((

i3laze, 2022-01-04:
Just wondering which countries are affected.
I’m from Russia, where government filters traffic from time to time. (VPN didn’t help me much).

XDimik, 2022-01-04:
> Just wondering which countries are affected. I’m from Russia, where government filters traffic from time to time. (VPN didn’t help me much).

Me too, apparently these are the first bells about cheburnet((

RavenSystem, 2022-01-04:
You can not update over the air an old device with GitHub, because running firmware has the issue.

XDimik, 2022-01-04:
It's working!))

i3laze, 2022-01-04:
> It's working!))

Please elaborate on this short message.

XDimik, 2022-01-04:
> > It's working!))
> 
> Please elaborate on this short message. What? At what conditions?

At the last beta, after a large number of reboots, the device appeared in the home kit

RavenSystem, 2022-01-04:
Fixed in stable released.

https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_9.5.2

closed 2022-01-04 by RavenSystem