RavenSystem/esp-homekit-devices #1672: Enter setup mode

willliamchan, 2021-12-28:
Hi there

I am not very sure why but the 8 times power toggle never work with my two HAA 9.5.1 device, is there any other way to enter setup mode again without re-loading the firmware?

Million thanks!

W

tonysprenk, 2021-12-28:
https://apps.apple.com/nl/app/haa-home-manager-for-homekit/id1556105121?l=en

willliamchan, 2021-12-28:
Yes I got the app too and unfortunately the old WiFi no longer exists, any other suggestion please?

tonysprenk, 2021-12-28:
If the WiFi doesn’t exist the device should automatically go into AP mode after reboot. So reboot the device (unplug from power and plug back in) and then connect your phone or computer to the HAA-XXXX network. For future reference: please only use the issue section for faults with the firmware. The discussion section is meant for user questions.

willliamchan, 2021-12-28:
Gotcha. Sorry about that.

P.S. unfortunately I can’t see the HAA AP either, must be something to do with the device itself.

_____________________________________________

​William Chan

HK +852 9661 8686

UK +44 7898 888686

***@***.***

tonysprenk, 2021-12-28:
Have you tried [emergency setup mode](https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#entering-setup-mode) ?

willliamchan, 2021-12-28:
Yes I did, hotspot appear from network list for about a min but never able to connect.

Restarting the phone and/or using another phone are the same.

_____________________________________________

​William Chan

HK +852 9661 8686

UK +44 7898 888686

***@***.***

i3laze, 2021-12-29:
Actually, even in emergency setup mode  device respects WiFi settings and tries to connect to your WiFi access point.
Either turn off your WiFi AP, or check DHCP logs for  device IP address.. 
Then try http://ip:4567

willliamchan, 2021-12-29:
After several failures of joining HAA network and it worked.

Thank you very much!

_____________________________________________

​William Chan

HK +852 9661 8686

UK +44 7898 888686

***@***.***