RavenSystem/esp-homekit-devices #1718: flash fullhaaboot.bin file, while IOS Family app cannot find ESP devices

macleia, 2022-02-02:
1. download the latest fullhaaboot.bin, 
2. using esphome-flasher1.2.0 on Mac OS flash bin file into ESP-01S. 
3. Find Wifi HAA-XXXXXX, connect it.
4. Open 192.168.4.1:4567 configure WiFi network (do not config JSON) and save, page returns "OK"
5. esphome-flasher monitor whole process (see Logs file attached at the end)
6. 7-8 min later, log shows done, and configure JSON with code
{"c":{"1":0,"n":"IT_Commander_Temp Hum"),"a":[{"t":24,"g":2,"n":1,"j":30,"z":-4,"h":-0.1}]}
7. open family app on iphone, but cannot find any device :(

[Install_Logs.txt](https://github.com/RavenSystem/esp-homekit-devices/files/7987791/Install_Logs.txt)

RavenSystem, 2022-02-02:
Use a valid JSON. "h" can not be declared with a -0.1 value.

macleia, 2022-02-02:
thx, so do I correct it into an int number or just delete it? like this
{"c":{"1":0,"n":"IT_Commander_Temp Hum"),"a":[{"t":24,"g":2,"n":1,"j":30,"z":-4}]}

while from my perspective, Home app should find this device without JSON configure, am I right?

macleia, 2022-02-02:
I delete "h":-0.1, while the problem is still same, Home app cannot find device  :( @RavenSystem

macleia, 2022-02-02:
> Use a valid JSON. "h" can not be declared with a -0.1 value.
@RavenSystem
thx, so do I correct it into an int number or just delete it? like this
{"c":{"I":0,"n":"IT_Commander_Temp Hum"),"a":[{"t":24,"g":2,"n":1,"j":30,"z":-4}]}

from my perspective, Home app should find this device without JSON configure, am I right?

after a test with JSON : {"c":{"I":0,"n":"IT_Commander_Temp Hum"),"a":[{"t":24,"g":2,"n":1,"j":30,"z":-4}]}
the problem is same, Home app still cannot find device. :(