RavenSystem/esp-homekit-devices #1725: Can't connect to http server

Ov1diy, 2022-02-04:
After successful flashing of NodeMCU board, web server does't open... A wi-fi access point appears, in logs you can see that connecting is ok and server is running, but nothing happens after connecing to it. I have tried different phones and computers, different esp boards and flashers many times, but there is one result. What could be wrong? Sorry if this is a common problem, but after month of trying (even buying another esps), i still haven't found anything, that could help :(

[image]

tonysprenk, 2022-02-04:
You connect your account phone, computer to the AP and in the browser you go to: `http://192.168.4.1:4567/ ` and then what happens?

Ov1diy, 2022-02-04:
> You connect your phone, computer to the AP and in the browser you go to: `http://192.168.4.1:4567/ ` and then what happens?

OMG -__-
I feel so dumb, it works! It seemed to me that this is kind of local IP, i didnt even try this... I've been waiting all this time for the web interface to open itself, i know what you think %) Thank you a lot!

closed 2022-02-04 by Ov1diy