RavenSystem/esp-homekit-devices #1728: Delay in Binary Outputs

lxy764139720, 2022-02-05:
hello, I want to switch two GPIOs with 1 second delay between them. For example, switch GPIO 0 from low level to high level first, and then switch  GPIO 2 from high level to low level after 1 second. Is there anyway to do that? Thanks!

clint80, 2022-02-05:
hello I integrated a timer in the json of my garage door that once closed, actuate the security with
{"I": 5, "0": {"M": [[3,1]]}}. At the end of 5s Change Status of my 3rd relay

i3laze, 2022-02-15:
You should have add a dummy inching accessory and use internal notifications.
GPIO1 high -> notify 1sec inching switch on
Inching switch off -> notify GPIO2

@RavenSystem , I feel there’s a demand for delayed actions, can you introduce some delays for actions?

RavenSystem, 2022-02-19:
Use a switch outside HK to use delayed actions.

RavenSystem, 2022-02-19:
Please, use Discussions section and leave Issues section only for issues. Thanks!

https://github.com/RavenSystem/esp-homekit-devices/wiki#issues