RavenSystem/esp-homekit-devices #1729: Double motion sensor configuration strange behaviour

sorriso93, 2022-02-06:
**Detailed steps to reproduce it**
I have built a Wemos D1 Mini + DHT11 + two motion sensors.
Same configuration with one motion sensor is ok, adding the second and debugging I see a GPIO 99 not configured (was GPIO14 on json) and one motion sensor on homekit never gets triggered, nonetheless on the debug I get movement on GPIO99 that is the second sensor I bet (but not reported on GPIO14 as per the configuration)
Maybe something wrong on json config?

**JSON String**
{"c":{"n":"MOTTEMPBMAX","o":1},"a":[{"t":12,"f1":[{"g":0,"p":0}],"s":0,"i":5},{"t":12,"s":0,"i":5,"f1":[{"g":14,"p":0}]},{"t":24,"b":[],"g":2,"n":2,"z":-5,"h":10,"y0":[{"0":{"h":[{"h":"192.168.0.7","p":8086,"m":2,"u":"write?db=haadb","c":"sensor,location=bagnomax temperature=#HAA@0000,humidity=#HAA@0001"}]},"v":0,"r":1}]},{"1":{"s":[{"a":1}]},"i":0.5,"b":[],"s":0}]} 
**Used device**
Wemos D1 Mini + DHT11 + two motion sensors.
**HAA version**
10.2.0
**iOS/iPadOS version**
iOs 15.3 and 15.4
**Logs**

Jan 01 00:12:03 sensor,location=bagnomax temperature=23.4,humidity=34
Jan 01 00:12:03 <3> Net Action 192.168.0.7:8086 done
Jan 01 00:12:03 [11] Send Ev
Jan 01 00:12:03 [7] Send Ev
Jan 01 00:12:05 <1> DigI Sensor 1 GPIO 0
Jan 01 00:12:05 <1> Exec Action 1
Jan 01 00:12:05 [11] Send Ev
Jan 01 00:12:05 [7] Send Ev
Jan 01 00:12:10 <1> AutoOff
Jan 01 00:12:10 <1> DigI Sensor 0 GPIO 99
Jan 01 00:12:10 <1> Exec Action 0
Jan 01 00:12:10 [11] Send Ev
Jan 01 00:12:10 [7] Send Ev
Jan 01 00:12:10 [7] Get Ch
Jan 01 00:12:10 [7] Get Ch
Jan 01 00:12:10 [7] Get Ch
Jan 01 00:12:10 [7] Get Ch
Jan 01 00:12:12 <1> DigI Sensor 1 GPIO 0
Jan 01 00:12:12 <1> Exec Action 1
Jan 01 00:12:12 [11] Send Ev
Jan 01 00:12:12 [7] Send Ev
**Screen-shots or videos showing issue**

sorriso93, 2022-02-06:
Update, I tried also a simple configuration, seems that GPIO14 is not good as json config

Jan 01 00:00:00 

Jan 01 00:00:00 
Jan 01 00:00:00 Home Accessory Architect v10.2.0
Jan 01 00:00:00 (c) 2018-2021 José Antonio Jiménez Campos

Jan 01 00:00:00 
Jan 01 00:00:00 NORMAL

Jan 01 00:00:00 
Jan 01 00:00:00 JSON:
Jan 01 00:00:00  {"c":{"m":5,"n":"TEST","o":1},"a":[{"t":12,"f1":[{"g":14,"p":1}],"i":3,"s":0},{"1":{"s":[{"a":1}]},"i":0.5,"b":[],"s":0}]}

Jan 01 00:00:00 
Jan 01 00:00:00 Hostname TEST
Jan 01 00:00:00 Ping period 5 secs
Jan 01 00:00:00 Setup mode time 5 secs
Jan 01 00:00:00 HK Server Clients 20
Jan 01 00:00:00 Unsecure connections 0
Jan 01 00:00:00 mDNS TTL 4500
Jan 01 00:00:00 Toggles to enter setup 8
Jan 01 00:00:00 <0> Exec Action 0

Jan 01 00:00:00 
Jan 01 00:00:00 ** ACCESSORY 1

Jan 01 00:00:00 
Jan 01 00:00:00 * SERVICE 1
Jan 01 00:00:00 Type 12: BINARY SENSOR
Jan 01 00:00:00 KS main 1 child 1
Jan 01 00:00:00 Serial F445D9-1
Jan 01 00:00:00 Exec Ac on Boot 1
Jan 01 00:00:00 New Input GPIO 14: pullup 1, inv 0, filter 0, mode 0
Jan 01 00:00:00 New DigI GPIO 14: type 1, status 1
Jan 01 00:00:00 <1> DigI Sensor 1 GPIO 99
Jan 01 00:00:00 <1> Exec Action 1
Jan 01 00:00:00 Free Heap 41360

Jan 01 00:00:00 
Jan 01 00:00:00 ** ACCESSORY 2

Jan 01 00:00:00 
Jan 01 00:00:00 * SERVICE 2
Jan 01 00:00:00 Type 1: SWITCH / OUTLET
Jan 01 00:00:00 KS main 1 child 1
Jan 01 00:00:00 Serial F445D9-2
Jan 01 00:00:00 New System Action 1: val 1
Jan 01 00:00:00 Exec Ac on Boot 1
Jan 01 00:00:00 Init State 0
Jan 01 00:00:00 Set init state
Jan 01 00:00:00 <2> Setter ON 0
Jan 01 00:00:00 Setup mode trigger -127/8
Jan 01 00:00:00 <2> Exec Action 0
Jan 01 00:00:00 Free Heap 40848

Jan 01 00:00:00 
Jan 01 00:00:00 Device Category: 10

Jan 01 00:00:00 
Jan 01 00:00:01 Wifi config init
Jan 01 00:00:01 <0> Exec Action 1
Jan 01 00:00:01 Saved BSSID: c0c9e39cd086
Jan 01 00:00:01 Wifi Mode: Roaming
Jan 01 00:00:01 Start Wifi smart connect scan
Jan 01 00:00:01 Set hostname
Jan 01 00:00:02 Disarming Emergency Setup Mode
Jan 01 00:00:03 <1> AutoOff
Jan 01 00:00:03 <1> DigI Sensor 0 GPIO 99
Jan 01 00:00:03 <1> Exec Action 0
Jan 01 00:00:05 Saving states
Jan 01 00:00:05 scandone
Jan 01 00:00:05 Searching best BSSID for MAXNETHOME
Jan 01 00:00:05 RSSI -59, Ch 5 - c0c9e39cc9a6
Jan 01 00:00:05 RSSI -59, Ch 5 - c0c9e39cd102
Jan 01 00:00:05 RSSI -44, Ch 5 - c0c9e39cd086
Jan 01 00:00:05 Best BSSID: c0c9e39cd086
Jan 01 00:00:06 scandone
Jan 01 00:00:08 reconnect
Jan 01 00:00:09 scandone
Jan 01 00:00:10 add 0
Jan 01 00:00:10 aid 18
Jan 01 00:00:10 cnt 

Jan 01 00:00:10 
Jan 01 00:00:10 connected with MAXNETHOME, channel 5
Jan 01 00:00:10 dhcp client start...
Jan 01 00:00:12 ip:192.168.0.155,mask:255.255.255.0,gw:192.168.0.3
Jan 01 00:00:12 Free Heap 45924
Jan 01 00:00:12 Starting HKServer
Jan 01 00:00:12 Using existing HomeKit ID: F7:41:BF:AA:6C:8D
Jan 01 00:00:12 Found admin pairing with 45B645D6-3893-47DE-9C72-9AFF3B613D2C, disabling pair setup
Jan 01 00:00:12 Configuring mDNS
Jan 01 00:00:12 Free Heap 36068
Jan 01 00:00:12 >>> mDNS probing 1
Jan 01 00:00:12 mDNS announcement: Name=HAA-F445D9 md=RavenSystem HAA Kestrelpv=1.1id=F7:41:BF:AA:6C:8Dc#=1s#=1ff=0sf=0ci=10sh=i5bhyA== Port=5556 TTL=4500
Jan 01 00:00:13 <0> Exec Action 2
Jan 01 00:00:14 NTP upd (-1)
Jan 01 00:00:26 NTP upd (-1)
Jan 01 00:00:26 >>> mDNS probing 2
Jan 01 00:00:33 >>> mDNS working TTL 4500
Jan 01 00:00:38 NTP upd (-1)
Jan 01 00:00:47 [3] New 192.168.0.83:39393 (1/20) Free HEAP: 33312
Jan 01 00:00:47 [3] Verify 1/2
Jan 01 00:00:48 [3] Verify 2/2
Jan 01 00:00:48 [3] Pairing with 45B645D6-3893-47DE-9C72-9AFF3B613D2C
Jan 01 00:00:48 [3] Verify OK
Jan 01 00:00:48 [3] Get Acc
Jan 01 00:00:48 [3] Get Ch
Jan 01 00:00:48 [3] Get Ch
Jan 01 00:00:48 [3] Get Ch
Jan 01 00:00:48 [3] Get Ch
Jan 01 00:00:48 [3] Upd Ch
Jan 01 00:00:48 [3] Upd Ch
Jan 01 00:00:48 [3] Get Ch
Jan 01 00:00:50 NTP upd (-1)

Hanimaniac, 2022-02-14:
I have exact the same problem and no idea how to change it.

sorriso93, 2022-02-18:
It seems I solved it was an hardware problem using the wemos D1 PIR shield. If you change the pin by soldering on another pin configuration this is effective only using the board pin and not via the cable connected to the other shiel
[image]
d
So basically speaking I had tow PIRs sending signal on same GPIO in my case

sorriso93, 2022-02-18:
I was optimistic. Now the problem remains that both PIR randomly becomes active (even with no movement)

Jan 01 00:00:02 Wifi config init
Jan 01 00:00:02 <0> Exec Action 1
Jan 01 00:00:02 Saved BSSID: c0c9e39cd086
Jan 01 00:00:02 Wifi Mode: Roaming
Jan 01 00:00:02 Start Wifi smart connect scan
Jan 01 00:00:02 Set hostname
Jan 01 00:00:02 Disarming Emergency Setup Mode
Jan 01 00:00:03 <3> Starting TH sensor
Jan 01 00:00:03 <3> TH sensor
Jan 01 00:00:03 <3> TEMP 19.6C
Jan 01 00:00:03 <3> Wilcard 0 0.00
Jan 01 00:00:03 <3> Exec Action 51
Jan 01 00:00:03 <3> HUM 40
Jan 01 00:00:05 <1> AutoOff
Jan 01 00:00:05 <1> DigI Sensor 0 GPIO 99
Jan 01 00:00:05 <1> Exec Action 0
Jan 01 00:00:05 <2> AutoOff
Jan 01 00:00:05 <2> DigI Sensor 0 GPIO 99
Jan 01 00:00:05 <2> Exec Action 0
Jan 01 00:00:05 Saving states
Jan 01 00:00:06 scandone
Jan 01 00:00:06 Searching best BSSID for MAXNETHOME
Jan 01 00:00:06 RSSI -62, Ch 5 - c0c9e39cc9a6
Jan 01 00:00:06 RSSI -42, Ch 5 - c0c9e39cd086
Jan 01 00:00:06 RSSI -67, Ch 5 - c0c9e39cd102
Jan 01 00:00:06 Best BSSID: c0c9e39cd086
Jan 01 00:00:07 scandone
Jan 01 00:00:08 <2> DigI Sensor 1 GPIO 14
Jan 01 00:00:08 <2> Exec Action 1
Jan 01 00:00:08 <1> DigI Sensor 1 GPIO 0
Jan 01 00:00:08 <1> Exec Action 1
Jan 01 00:00:09 reconnect
Jan 01 00:00:09 scandone
Jan 01 00:00:10 add 0
Jan 01 00:00:10 aid 15
Jan 01 00:00:10 cnt 

Jan 01 00:00:11 
Jan 01 00:00:11 connected with MAXNETHOME, channel 5
Jan 01 00:00:11 dhcp client start...
Jan 01 00:00:12 ip:192.168.0.204,mask:255.255.255.0,gw:192.168.0.3
Jan 01 00:00:12 Free Heap 44452
Jan 01 00:00:12 Starting HKServer
Jan 01 00:00:12 Using existing HomeKit ID: 5F:F8:56:29:7E:97
Jan 01 00:00:12 Found admin pairing with 45B645D6-3893-47DE-9C72-9AFF3B613D2C, disabling pair setup
Jan 01 00:00:12 Configuring mDNS
Jan 01 00:00:12 Free Heap 34592
Jan 01 00:00:13 <2> AutoOff
Jan 01 00:00:13 <2> DigI Sensor 0 GPIO 99
Jan 01 00:00:13 <2> Exec Action 0
Jan 01 00:00:13 <1> AutoOff
Jan 01 00:00:13 <1> DigI Sensor 0 GPIO 99
Jan 01 00:00:13 <1> Exec Action 0
Jan 01 00:00:13 >>> mDNS probing 1
Jan 01 00:00:13 mDNS announcement: Name=HAA-7EAC56 md=RavenSystem HAA Kestrelpv=1.1id=5F:F8:56:29:7E:97c#=1s#=1ff=0sf=0ci=10sh=D+09EA== Port=5556 TTL=4500
Jan 01 00:00:13 <0> Exec Action 2
Jan 01 00:00:13 [5] New 192.168.0.117:13776 (1/20) Free HEAP: 28456
Jan 01 00:00:13 [5] Verify 1/2
Jan 01 00:00:13 [6] New 192.168.0.174:45257 (2/20) Free HEAP: 25856
Jan 01 00:00:13 [6] Verify 1/2
Jan 01 00:00:14 [5] Verify 2/2
Jan 01 00:00:14 [5] Pairing with 45B645D6-3893-47DE-9C72-9AFF3B613D2C
Jan 01 00:00:14 [5] Verify OK
Jan 01 00:00:14 [6] Verify 2/2
Jan 01 00:00:14 [6] Pairing with 45B645D6-3893-47DE-9C72-9AFF3B613D2C
Jan 01 00:00:14 [6] Verify OK
Jan 01 00:00:14 [5] Get Acc
Jan 01 00:00:14 NTP upd (-1)
[... 283 lines cut ...]
Jan 01 00:00:18 [3] Verify 1/2
Jan 01 00:00:19 [3] Verify 2/2
Jan 01 00:00:19 [3] Pairing with 45B645D6-3893-47DE-9C72-9AFF3B613D2C
Jan 01 00:00:19 [3] Verify OK
Jan 01 00:00:19 [3] Get Acc
Jan 01 00:00:19 [3] Get Ch
Jan 01 00:00:19 [3] Get Ch
Jan 01 00:00:19 [3] Get Ch
Jan 01 00:00:19 [3] Get Ch
Jan 01 00:00:19 <2> AutoOff
Jan 01 00:00:19 <2> DigI Sensor 0 GPIO 99
Jan 01 00:00:19 <2> Exec Action 0
Jan 01 00:00:19 <1> AutoOff
Jan 01 00:00:19 <1> DigI Sensor 0 GPIO 99
Jan 01 00:00:19 <1> Exec Action 0
Jan 01 00:00:19 [3] Get Ch
Jan 01 00:00:19 [3] Upd Ch
Jan 01 00:00:19 [3] Upd Ch
Jan 01 00:00:19 [3] Upd Ch
Jan 01 00:00:19 [3] Upd Ch
Jan 01 00:00:19 [3] Get Ch
Jan 01 00:00:20 [3] Get Ch
Jan 01 00:00:20 >>> mDNS probing 2
Jan 01 00:00:20 [3] Get Ch
Jan 01 00:00:23 <2> DigI Sensor 1 GPIO 14

sorriso93, 2022-02-18:
Maybe the problem is that both accessories have a Digi on GPIO99? What is Digi on GPIO99, I didn't declared it in the json....

sorriso93, 2022-02-18:
I also tried the json from [this old issue](https://github.com/RavenSystem/esp-homekit-devices/issues/856) that seems ok but in my case has the same behaviour

RavenSystem, 2022-02-19:
GPIO99 is showed when input is triggered by internal mechanism, like an inching time, or a service notification. There is not any bug.

RavenSystem, 2022-02-19:
Please, use Discussions section and leave Issues section only for issues. Thanks!

https://github.com/RavenSystem/esp-homekit-devices/wiki#issues