RavenSystem/esp-homekit-devices #1730: Can

qubiack, 2022-02-06:
When I open a setup page I don't see any version to update as on the presented image and I can't update my Shelly 1 OTA. I tried to use also HAA Manager from AppStore but both - update devices & update all devices have the same issue and I can't update them.

[image]

i3laze, 2022-02-15:
I guess it’s your old OTA doesn’t support recent GitHub changes, like partial download.

Please try my custom HTTP server, which should allow you to bump to OTA v5.8.2:
91.218.86.30:8888

RavenSystem, 2022-02-19:
Please, use Discussions section and leave Issues section only for issues. Thanks!

https://github.com/RavenSystem/esp-homekit-devices/wiki#issues