RavenSystem/esp-homekit-devices #1761: Suggested improvement to textarea html - turns off autocorrect

andyallcorn, 2022-02-27:
Certain keys in JSON configurations -- "i" being a common one, as well as smart quotes -- are very susceptible to being broken by autocorrect on iOS and macOS. The attached diff would:

1. Turn off various kinds of autocorrect, meaning words will not be suggested/capitalised and smart quotes will not be used.
2. Make the texture a little larger so more JSON can be seen and edited.

[textarea.txt](https://github.com/RavenSystem/esp-homekit-devices/files/8147789/textarea.txt)

RavenSystem, 2022-02-27:
Many thanks! It will be available in next version.

closed 2022-02-27 by RavenSystem

Daveblanche, 2022-02-27:
Marvellous idea!
Autocorrect is the bane of our lives.

i3laze, 2022-03-07:
Does this fix broken “ quotes also?
Quotes in iOS Safari are the wrong ones.