RavenSystem/esp-homekit-devices #1792: Failure in downloading otamain.bin.sec

law1618, 2022-03-11:
After I configure the wifi ssid and password, the esp device connects to the Wifi and start downloading the necessary installation document. However, it stuck at otamain.bin.sec. Below please find the repeated log. 

*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECT
DNS..!
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECT
DNS..!
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECT
DNS..!
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECT
DNS..!
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECT
DNS..!
! Tries 1
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECT
DNS..OK
Socket..OK
Connect..!
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECT
DNS..OK
Socket..OK
Connect..!
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECT
DNS..OK
Socket..OK
Connect..!
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECT
DNS..OK
Socket..OK
Connect..!
*** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
*** CONNECT
DNS..OK
Socket..OK

law1618, 2022-03-18:
@RavenSystem Can you please help? I tried to use custom OTA update server but still no luck. 

Thanks a lot in advance!

swpolgla, 2022-04-11:
I have the exact same issue after flashing a Gosund WP3 OTA with tuya convert.

```
Advanced ESP Logger (c) 2022 José A. Jiménez Campos

0008.733 192.168.1.89-BA263A: 
0008.734 192.168.1.89-BA263A: 
0008.734 192.168.1.89-BA263A: HAA Installer v6.0.7
0008.734 192.168.1.89-BA263A: 
0008.734 192.168.1.89-BA263A: 
0008.740 192.168.1.89-BA263A: REPO https://github.com/RavenSystem/haa/releases/latest/download 443
0008.746 192.168.1.89-BA263A: Current HAA v0.0.0
0008.746 192.168.1.89-BA263A: 
0008.747 192.168.1.89-BA263A: INIT
0010.745 192.168.1.89-BA263A: 
0010.746 192.168.1.89-BA263A: *** STARTING
0010.746 192.168.1.89-BA263A: 
0010.747 192.168.1.89-BA263A: 
0010.747 192.168.1.89-BA263A: HAABOOT
0010.747 192.168.1.89-BA263A: 
0010.747 192.168.1.89-BA263A: >>> Get sign
0010.748 192.168.1.89-BA263A: *** DOWNLOADING
0010.748 192.168.1.89-BA263A: *** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
0010.749 192.168.1.89-BA263A: *** CONNECT
0010.960 192.168.1.89-BA263A: DNS..OK
0010.961 192.168.1.89-BA263A: Socket..OK
0020.203 192.168.1.89-BA263A: Connect..!
0020.205 192.168.1.89-BA263A: *** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
0020.205 192.168.1.89-BA263A: *** CONNECT
0020.206 192.168.1.89-BA263A: DNS..OK
0020.206 192.168.1.89-BA263A: Socket..OK
0029.703 192.168.1.89-BA263A: Connect..!
0029.705 192.168.1.89-BA263A: *** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
0029.705 192.168.1.89-BA263A: *** CONNECT
0029.706 192.168.1.89-BA263A: DNS..OK
0029.706 192.168.1.89-BA263A: Socket..OK
0029.703 192.168.1.89-BA263A: Connect..!
0029.705 192.168.1.89-BA263A: *** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
0029.705 192.168.1.89-BA263A: *** CONNECT
0029.706 192.168.1.89-BA263A: DNS..OK
0029.706 192.168.1.89-BA263A: Socket..OK
0039.203 192.168.1.89-BA263A: Connect..!
0039.205 192.168.1.89-BA263A: *** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
0039.205 192.168.1.89-BA263A: *** CONNECT
0039.203 192.168.1.89-BA263A: Connect..!
0039.205 192.168.1.89-BA263A: *** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
0039.205 192.168.1.89-BA263A: *** CONNECT
0040.243 192.168.1.89-BA263A: DNS..!
0040.244 192.168.1.89-BA263A: *** FORWARDING: github.com:443/RavenSystem/haa/releases/latest/download/otamain.bin.sec
...
```

closed 2022-05-05 by stale[bot]