RavenSystem/esp-homekit-devices #1796: iOS 15.4 - window covering siri bug?

asbesto01, 2022-03-17:
**Detailed steps to reproduce it**

1) Download iOS 15.4.
2) Try to ask Siri to open a window blind / shades / whatever you call it.
3)

Each iOS 15.4 device will not be able to perform any action
Each iOS 15.3.1 device (or previous version) will be able to perform action

**JSON String**
{"c":{"l":13,"b":[{"g":10,"t":5}]},"a":[{"t":45,"o":22,"c":22,"f":0,"0":{"r":[{"g":12,"v":1},{"g":5}]},"1":{"r":[{"g":12},{"g":5,"v":1}]},"2":{"r":[{"g":12},{"g":5}]},"3":{"r":[{"g":12,"i":1},{"g":5}]},"4":{"r":[{"g":12},{"g":5,"i":1}]},"f3":[{"g":0}],"f4":[{"g":9}]}]}

**Used device**
Sonoff dual r2

**HAA version**
10.4.3 Kestrel

**iOS/iPadOS version**
15.4 - iPhone 11 Pro Max, iPhone XS, iPhone 11, iPad Pro, seem all affected.

**Logs**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)

**Screen-shots or videos showing issue**

Kienz, 2022-03-17:
It's a Siri bug. @RavenSystem can't do nothing.
https://www.iphone-ticker.de/siri-will-unter-ios-15-4-keine-rolllaeden-mehr-steuern-187702/

asbesto01, 2022-03-17:
thanks, on which device are you experiencing this?

Kienz, 2022-03-17:
iPhone 13 Pro, iPad Pro, macOS, HomePod mini, Apple TV (all with latest software)

asbesto01, 2022-03-17:
no sorry, I mean which devices are you using to control your blinds?

I use a sonoff dual r2, you?

asbesto01, 2022-03-17:
ok so non native homekit device. It seems that native homekit devices are not having this issue?

https://www.reddit.com/r/HomeKit/comments/te7s4u/ios_154_siri_no_longer_understands_how_to_operate/

that's why I escalated to you guys :)

Kienz, 2022-03-17:
I think Siri was updated. Please try again. For me it works again.

asbesto01, 2022-03-17:
tried right now, not working at all.

jubudu, 2022-03-18:
Same issue here: Using HAA and an ESP8266 does not work with Siri since updating to IOS 15.4. 
I assume, it is not a Siri bug, cause two other blinds are controlled by a Raspberry pi zero W running homekit and can still be controlled with Siri.

xrust83, 2022-03-18:
Russian Siri, when asked to open a window at 80%, answers that nothing is playing now, or an error occurred when changing the volume.

asbesto01, 2022-03-19:
I discovered that:

If you say “Siri close the blinds” it’s not working
If you say “siri set blinds position to 0%” it’s closing the blinds.

you have to include word POSITION in the request, please try

asbesto01, 2022-03-24:
> Yes, with Position it works. But that is very uncomfortable! It should work with „Close/Open blind“. As mentioned above, using a raspberry with homebridge still works with open and close Siri command.

the fact this works using raspberry with homebridge makes me think this could be a raven core issue?

@RavenSystem any help José? muchas gracias!

Kienz, 2022-04-01:
15.4.1 should fix this.

RavenSystem, 2022-04-01:
This is an Apple bug, not HAA.