RavenSystem/esp-homekit-devices #1797: Lock Mechanism State Feedback

yuanxun-yx, 2022-03-18:
I'm adding HomeKit support to a type of locks that is very common in China. I used a very similar JSON file as the one provided in [Wiki](https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#state-input-example-1). The state of the lock is provided by a GPIO pin, which indicates whether the lock is locked. I intended to get feedback of the lock by doing so, but it didn't seem to report error when the lock is not correctly unlocked when I sent an unlock signal. (I know it's kind of a feature request, sorry.)
```
{
  "c": { "l": 13 },
  "a": [{
    "t": 4,
    "0": { "r": [{ "g": 12, "v": 0 }] },
    "1": { "r": [{ "g": 12, "v": 1 }] },
    "f0": [{ "g": 14, "t": 1 }],
    "f1": [{ "g": 14, "t": 1 }]
  }]
}
```

yuanxun-yx, 2022-04-09:
It’s still not resolved.

KagurazakaSarah, 2022-04-10:
I have the same problem.

closed 2022-05-05 by stale[bot]