RavenSystem/esp-homekit-devices #1804: Build from source error

OwenL1, 2022-03-27:
Hi there, 

Thank you for the hard works! I am trying to build from source code following the instruction from the Build instruction page, I got this error, does anyone know how to solve this?

SDK processing stage 2: Renaming symbols in SDK library build/sdklib/libwpa_stage1.a -> build/sdklib/libwpa.a
LD build/haamain.out
./build/sdklib/libwpa.a(wpa_main.o): In function `sdk_eagle_auth_done':
(.irom0.text+0x295): dangerous relocation: call0: call target out of range: sdk_eagle_auth_done
collect2: error: ld returned 1 exit status
make[1]: *** [build/haamain.out] Error 1
make: *** [rebuild] Error 2

Thanks in advance.

lee238, 2022-03-31:
It‘s sloved by:
https://github.com/SuperHouse/esp-open-rtos/issues/674#issuecomment-439717688

closed 2022-04-24 by stale[bot]

LouisLee985, 2024-03-29:
esptool.py v1.3
/opt/xtensa-lx106-elf/bin/esptool.py:1233: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0, use inspect.signature() or inspect.getfullargspec()
  operation_args,_,_,_ = inspect.getargspec(operation_func)