RavenSystem/esp-homekit-devices #1806: Shelly dimmer 2 not working after update 10.4.4

Alexanco8, 2022-03-30:
**Detailed steps to reproduce it**

Hi, updated shelly dimmer 2 from 10.4 to 10.4.4 and now I can’t switch off or dimmer the light, tried to reset homekit and replace the json but no way. Light work at 100% and when try to switch off it decrease around to 50%. Could you help?

**JSON String**

{"c":{"f":20,"r":[{"n":0}],"l":15,"i":0,"b":[{"g":13,"t":5}]},"a":[{"t":30,"b":[{"g":13},{"g":12,"i":1,"p":0,"t":0},{"g":14,"i":1,"p":0},{"g":14,"t":0}],"0":{"u":[{"v":"010202060000f401000000ff04"}]},"y0":[{"v":1,"0":{"u":[{"v":"010202060a00f4010000010904"}]}},{"v":5,"0":{"u":[{"v":"010202063200f4010000013104"}]}},{"v":10,"0":{"u":[{"v":"010202066400f4010000016304"}]}},{"v":15,"0":{"u":[{"v":"010202069600f4010000019504"}]}},{"v":20,"0":{"u":[{"v":"01020206c800f401000001c704"}]}},{"v":25,"0":{"u":[{"v":"01020206fa00f401000001f904"}]}},{"v":30,"0":{"u":[{"v":"010202062c01f4010000012c04"}]}},{"v":35,"0":{"u":[{"v":"010202065e01f4010000015e04"}]}},{"v":40,"0":{"u":[{"v":"010202069001f4010000019004"}]}},{"v":45,"0":{"u":[{"v":"01020206c201f401000001c204"}]}},{"v":50,"0":{"u":[{"v":"01020206f401f401000001f404"}]}},{"v":55,"0":{"u":[{"v":"010202062602f4010000012704"}]}},{"v":60,"0":{"u":[{"v":"010202065802f4010000018b04"}]}},{"v":65,"0":{"u":[{"v":"010202068a02f4010000010904"}]}},{"v":70,"0":{"u":[{"v":"01020206bc02f401000001bd04"}]}},{"v":75,"0":{"u":[{"v":"01020206ee02f401000001ef04"}]}},{"v":80,"0":{"u":[{"v":"010202060032f4010000013104"}]}},{"v":85,"0":{"u":[{"v":"010202065203f4010000015404"}]}},{"v":90,"0":{"u":[{"v":"010202068403f4010000018604"}]}},{"v":95,"0":{"u":[{"v":"01020206b603f401000001b804"}]}},{"v":100,"0":{"u":[{"v":"01020206e803f401000001ea04"}]}}]}]}

**Used device**
Shelly dimmer 2
**HAA version**
10.4.4
**iOS/iPadOS version**
15.4
**Logs**

**Screen-shots or videos showing issue**

RavenSystem, 2022-03-31:
Try v10.4.5.

Alexanco8, 2022-03-31:
thank you for suggestion but still not working, tried updating to 10.4.5 but still not dimmering and flashing once turned off, tried also to change the json you changed in official wiki but same result, here below you find the previous json and the actual but result is the same. light goes only to 100% or low light flashing.

json previously working with 10.4

{"c":{"f":20,"r":[{"n":0}],"l":15,"i":0,"b":[{"g":13,"t":5}]},"a":[{"t":30,"b":[{"g":13},{"g":12,"i":1,"p":0,"t":0},{"g":14,"i":1,"p":0},{"g":14,"t":0}],"0":{"u":[{"v":"010202060000f401000000ff04"}]},"y0":[{"v":1,"0":{"u":[{"v":"010202060a00f4010000010904"}]}},{"v":5,"0":{"u":[{"v":"010202063200f4010000013104"}]}},{"v":10,"0":{"u":[{"v":"010202066400f4010000016304"}]}},{"v":15,"0":{"u":[{"v":"010202069600f4010000019504"}]}},{"v":20,"0":{"u":[{"v":"01020206c800f401000001c704"}]}},{"v":25,"0":{"u":[{"v":"01020206fa00f401000001f904"}]}},{"v":30,"0":{"u":[{"v":"010202062c01f4010000012c04"}]}},{"v":35,"0":{"u":[{"v":"010202065e01f4010000015e04"}]}},{"v":40,"0":{"u":[{"v":"010202069001f4010000019004"}]}},{"v":45,"0":{"u":[{"v":"01020206c201f401000001c204"}]}},{"v":50,"0":{"u":[{"v":"01020206f401f401000001f404"}]}},{"v":55,"0":{"u":[{"v":"010202062602f4010000012704"}]}},{"v":60,"0":{"u":[{"v":"010202065802f4010000018b04"}]}},{"v":65,"0":{"u":[{"v":"010202068a02f4010000010904"}]}},{"v":70,"0":{"u":[{"v":"01020206bc02f401000001bd04"}]}},{"v":75,"0":{"u":[{"v":"01020206ee02f401000001ef04"}]}},{"v":80,"0":{"u":[{"v":"010202060032f4010000013104"}]}},{"v":85,"0":{"u":[{"v":"010202065203f4010000015404"}]}},{"v":90,"0":{"u":[{"v":"010202068403f4010000018604"}]}},{"v":95,"0":{"u":[{"v":"01020206b603f401000001b804"}]}},{"v":100,"0":{"u":[{"v":"01020206e803f401000001ea04"}]}}]}]}

actual json in the wiki not working as the other one with 10.4.5

{"c":{"r":[{"n":0}],"l":16,"i":0,"b":[{"g":13,"t":5}]},"a":[{"t":30,"b":[{"g":13},{"g":12,"i":1,"p":0,"t":0},{"g":14,"i":1,"p":0},{"g":14,"t":0}],"0":{"u":[{"v":"010202060000f401000000ff04"}]},"y0":[{"v":1,"0":{"u":[{"v":"010202060a00f4010000010904"}]}},{"v":5,"0":{"u":[{"v":"010202063200f4010000013104"}]}},{"v":10,"0":{"u":[{"v":"010202066400f4010000016304"}]}},{"v":15,"0":{"u":[{"v":"010202069600f4010000019504"}]}},{"v":20,"0":{"u":[{"v":"01020206c800f401000001c704"}]}},{"v":25,"0":{"u":[{"v":"01020206fa00f401000001f904"}]}},{"v":30,"0":{"u":[{"v":"010202062c01f4010000012c04"}]}},{"v":35,"0":{"u":[{"v":"010202065e01f4010000015e04"}]}},{"v":40,"0":{"u":[{"v":"010202069001f4010000019004"}]}},{"v":45,"0":{"u":[{"v":"01020206c201f401000001c204"}]}},{"v":50,"0":{"u":[{"v":"01020206f401f401000001f404"}]}},{"v":55,"0":{"u":[{"v":"010202062602f4010000012704"}]}},{"v":60,"0":{"u":[{"v":"010202065802f4010000018b04"}]}},{"v":65,"0":{"u":[{"v":"010202068a02f4010000010904"}]}},{"v":70,"0":{"u":[{"v":"01020206bc02f401000001bd04"}]}},{"v":75,"0":{"u":[{"v":"01020206ee02f401000001ef04"}]}},{"v":80,"0":{"u":[{"v":"010202060032f4010000013104"}]}},{"v":85,"0":{"u":[{"v":"010202065203f4010000015404"}]}},{"v":90,"0":{"u":[{"v":"010202068403f4010000018604"}]}},{"v":95,"0":{"u":[{"v":"01020206b603f401000001b804"}]}},{"v":100,"0":{"u":[{"v":"01020206e803f401000001ea04"}]}}]}]}

RavenSystem, 2022-03-31:
Please, downgrade again to 10.4.0 and confirm that it works. Then, install 10.4.2 and 10.4.3 to find where is the issue (10.4.1 is not availbale).

https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#installing-a-specific-version

tokamac, 2022-04-08:
Have you tried to downgrade to 10.4.0 as RaveSystem told you to do, and what is the result with your JSON?

OwenL1, 2022-04-10:
> 

This is off the topic, but hope you can help me, I have brought the Shelly dimmer 2, I could not flash it using the usb to serial adaptor. I connected the GPIO0 and reset to ground, TX to RX, RX to TX. I keep getting error said no serial information received. I am using Mac and NodeMCU flasher software. Just wondering how did you flash the HAA in? Thank you in advance

closed 2022-05-05 by stale[bot]