RavenSystem/esp-homekit-devices #1809: Shelly 2.5 firmware flash

pyrotechnik, 2022-04-02:
Hi,
I use the latest HAA firmware on the sonoff mini R2 without any problems. I have a problem with flash execution on shelly 2.5. I connect GPIO0 to GND during startup, but I still can't connect to the shelly. I use ESP 01 and MacOS.

OwenL1, 2022-04-10:
I have the same problem on Shelly dimmer 2. Hope someone can help us here.

Fudalii, 2022-04-21:
You have new version Shelly with **ESP32**!! HAA not work with this chip, And you're probably writing about shelly 2P. 
I don't know anything about changing the WiFi chip to 2.5. 
More and more devices are changing ESP82xx to ESP32. Author, it's time to think about compiling for this chip :)

closed 2022-07-10 by stale[bot]