RavenSystem/esp-homekit-devices #1816: All devices stop work

richardlimaa, 2022-04-13:
**Detailed steps to reproduce it**
All my devices stop working with haa manager and home, even after update, anyone same problem?

richardlimaa, 2022-04-13:
[image]
[image]
[image]

Ucoz3000, 2022-04-16:
some devices are not available after flashing HAA_10.4.6
back to firmware HAA_10.4.5 devices available

richardlimaa, 2022-04-16:
There is a way to do it without remove from wall?

Ucoz3000, 2022-04-16:
Firmware can also be downgraded by placing an older copy of the files on your web server.

richardlimaa, 2022-04-17:
Happy Easter, and thanks for this, but can i use for example a raven webserver to downgrade once i dont have my own webserver?

Ucoz3000, 2022-04-18:
Local server XAMPP

richardlimaa, 2022-04-18:
[image]

richardlimaa, 2022-04-18:
I tried many versions but i cant manager randon firmwares

richardlimaa, 2022-04-18:
@
[image]

richardlimaa, 2022-04-18:
@`
[image]
`

richardlimaa, 2022-04-18:
[image]

.

HonzaJaros, 2022-04-19:
> **Detailed steps to reproduce it**
> All my devices stop working with haa manager and home, even after update, anyone same problem?

Same here as well, just waiting for new firmware and then I’ll flick switch twice, goes to setup (if broken booting up) and auto update….
Hopefully new version is gonna be soon

richardlimaa, 2022-04-24:
Everything is ok after 10.5.1, without change nothing thanks 🙏🏼

closed 2022-04-24 by richardlimaa