RavenSystem/esp-homekit-devices #1821: Imposible to enter setup mode

Tibi-rvb, 2022-04-19:
Used to have multiple Shelly RGBW2 with HAA but now moved apartments and I flashed a new RGBW2 with HAA and it's impossible to enter setup mode.
I added the RGBW2 to wifi, it's responding to ping but not opening the setup page ... even removing power does not work. 
Only way to enter setup is if I shut down the wifi router.

Tried also from HAA iPhone app, no luck ... any pointers ?

closed 2022-04-24 by Tibi-rvb

tokamac, 2022-05-23:
See https://github.com/yaourdt/mgos-to-tasmota/issues/73
I am in the same stateless annoyance since I tried to install HAA with this broken OTA software on two Shelly Dimmer 2 modules. They are on the local network (IPs visible in the router interface, answer ping), but they are not in HomeKit, are not in HAA setup mode, and do not answer Shelly's IP setting URL 192.168.33.1 anymore. I don't know what to do :(

xrust83, 2022-05-24:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#installing-haa

Ip:port  192.168.33.1:4567

tokamac, 2022-05-24:
Doesn't work. You're trying to combine the IP 192.168.33.1 used by Shelly's stock firmware with the port 4567 used by HAA setup mode, whereas:
- The two have nothing to do with each other
- There is no more stock firmware on the device, no more than HAA firmware, as yaourdt's software erased the stock FW without even verifying it can download the HAA FW first (which it cannot do). So there is no more firmware at all on the device.

This works however: https://github.com/yaourdt/mgos-to-tasmota/issues/73#issuecomment-1124240911
using SHDM-2.zip for the Dimmer 2.

xrust83, 2022-05-24:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#entering-setup-mode

tokamac, 2022-05-24:
I think you didn't understand the point of installing HAA through the 3rd-party OTA solution for Shelly modules. Maybe you should read the links I posted first and understand the whole context, and stop posting pointless messages on the content side, that are moreover barely understandable in English language on the form side.

RavenSystem, 2022-05-24:
https://github.com/RavenSystem/esp-homekit-devices/wiki#issues