RavenSystem/esp-homekit-devices #1827: JSON String

EgorMuhin, 2022-05-09:
Hi, please tell me what is wrong in the code, only the button is added to the three accessories (ESP8266)

{
  "c": {
    "o": 0
  },
  "a": [
    {
      "t": 3,
    "f0": [{ "g": 0, "t": 1 }],
    "f1": [{ "g": 0, "t": 2 }],
    "f2": [{ "g": 0, "t": 3 }]
  }]
},
    {
      "t": 5,
    "i": 10,
    "0": { "r": [{ "g": 16, "v": 0 }] },
    "1": { "r": [{ "g": 16, "v": 1 }] },
    "f1": [{ "g": 14 }]
  }]
},
    {
      "t": 55,
    "s": 5,
    "es":[{
      "i": 1,
      "1": { "m": [ [ -1, 4 ] ] }
    }]
  }]
}
  ]
}

Thanks:)

mariusvasss, 2022-05-09:
{
"c": {
"o": 0
},
"a": [
{
"t": 3,
"f0": [{ "g": 0, "t": 1 }],
"f1": [{ "g": 0, "t": 2 }],
"f2": [{ "g": 0, "t": 3 }]
}]
},
{
"t": 5,
"i": 10,
"0": { "r": [{ "g": 16, "v": 0 }] },
"1": { "r": [{ "g": 16, "v": 1 }] },
"f1": [{ "g": 14 }]
},
{
"t": 55,
"s": 5,
"es":[{
"i": 1,
"1": { "m": [ [ -1, 4 ] ] }
}]
}]
}
]
}

Should work

mariusvasss, 2022-05-09:
{
"c": {
"o": 0
},
"a": [
{
"t": 3,
"f0": [{ "g": 0, "t": 1 }],
"f1": [{ "g": 0, "t": 2 }],
"f2": [{ "g": 0, "t": 3 }]
},
{
"t": 5,
"i": 10,
"0": { "r": [{ "g": 16, "v": 0 }] },
"1": { "r": [{ "g": 16, "v": 1 }] },
"f1": [{ "g": 14 }]
}]
},
{
"t": 55,
"s": 5,
"es":[{
"i": 1,
"1": { "m": [ [ -1, 4 ] ] }
}]
}]
}
]
}

Sorry, this one.

mariusvasss, 2022-05-09:
{
"c": {
"o": 0
},
"a": [
{
"t": 3,
"f0": [{ "g": 0, "t": 1 }],
"f1": [{ "g": 0, "t": 2 }],
"f2": [{ "g": 0, "t": 3 }]
},
{
"t": 5,
"i": 10,
"0": { "r": [{ "g": 16, "v": 0 }] },
"1": { "r": [{ "g": 16, "v": 1 }] },
"f1": [{ "g": 14 }]
},
{
"t": 55,
"s": 5,
"es":[{
"i": 1,
"1": { "m": [ [ -1, 4 ] ] }
}]
}]
}
]
}

Sorry again.

EgorMuhin, 2022-05-10:
> {"c": {"o": 0},"а": [{"t": 3,"f0": [{ "g": 0, "t": 1 }],"f1": [{ "g": 0, "t": 2 }],"f2": [{ "g": 0, "t": 3 }]},{"t": 5,"i": 10,"0": { "r": [{ "g": 16, "v": 0 }] },"1": { "r": [{ "g": 16, "v": 1 }] },"f1": [{ "g": 14 }]},{"t": 55,"s": 5,"es":[{"i": 1,"1": { "m": [ [ -1, 4 ] ] }}]}]}]}
> 
> Еще раз извините.

Thank you very much, everything worked)

closed 2022-05-10 by EgorMuhin