RavenSystem/esp-homekit-devices #1829: No Wifi creation after flash

sgrqeaz, 2022-05-20:
Hi there, my issue might be quite annoying to ask. I flashed your .bin file using NodeMCU-PyFlasher onto my H801 Led Controller. The first time a wifi was created and i could see the setup page. I flashed it with Tasmota and then wanted to flash again with HAA (wiped data each time). With Tasmota wifi always appears but after reflashing with HAA only for HAA no wifi appears. I'm sorry this question is really strange, but I dont understand why Tasmota gives me a Wifi connection and HAA doesnt.

steini72, 2022-05-21:
To finish initial setup, click Save button and wait about 10 minutes until process finish (While installation is working, device doesn't show anything, and buttons don't work)

sgrqeaz, 2022-05-22:
> To finish initial setup, click Save button and wait about 10 minutes until process finish (While installation is working, device doesn't show anything, and buttons don't work)

i cannot setup, no wifi (HAA-----) appears after flash and connecting to power.

xrust83, 2022-05-22:
If you manage to flash in tasmot, try to flash through the file fullhaabut.bin. I shoot in Ukrainian, but you can set the automatic translation of subtitles, and the feeder series should help you. https://www.youtube.com/watch?v=mgYjDbohiT4&t=659s

stopar, 2022-05-23:
@sgrqeaz I have the same problem as well. After the first flash it the wifi works. Then it does not work after I've setup my wifi or put in the setup. It seems as if it bricked the board after trying to login into the default WIFI. 

Even after reflashing the .bin file again the default wifi (HAA-[MAC Address]) does not show the default Wifi. 

I am trying to find the old .bin file and see if I can get it to work.

xrust83, 2022-05-23:
It seems that no one reads the wiki. The access point appears to configure Wi-Fi, if you configured it when you first run it after the firmware, then why should it appear after the firmware? The dot will reappear only when you reset the wifi settings.
Stitched -> set up Wi-Fi and Json -> waited for 5-7 mm flash -> add the device to the homekit house
https://www.youtube.com/watch?v=O7lvkPdyyNY&t=150s

sgrqeaz, 2022-05-23:
if i fully wipe the chip and reflash the firmware, a wifi named HAA----- should still show up, but it doesnt. I have flashed other devices before with HAA and it worked perfectly, im just not sure why it won't work with my H801.

xrust83, 2022-05-23:
try to change different programming modes DIO, QIO, DOUT it has value for chips 8266 or 8285

stopar, 2022-05-23:
@sgrqeaz this worked for me: 

1. If you've configure the wifi the first time in Setup, the unit might have connected to your wifi already. 
2. The latest version of the .bin file blocks the Setup Menu after setting up wifi. 
3. Download [Home Manager App](https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Home-Manager)
4. Open the App and your device should be listed there. 
5. Click on the device > Setting > Setup. toggle it then wait until light flashing or something. 
6. At the bottom of the App you will see 'Setup Mode' change from Unreachable > Open in Safari (after light dance is finish in step #5). Once you click it will open the setup page in browser and you can configure it again.

sgrqeaz, 2022-05-23:
i bought a new controller, and now my flash works. Idk what exactly was wrong with the hardware but this did the trick. I will close this issue for now.

closed 2022-05-23 by sgrqeaz